kismetwireless/kismet #168: Kismet crashes on Raspberry Pi 3 B+ with newest Raspbian (10 - buster, kernel: 4.19.58-v7+)

JK2210, 2019-08-08:
I used Kismet successful under Raspbian Stretch with my Raspberry Pi 3 B+ for awhile.
Last day I switched to Raspbian Buster.
I installed Kismet with release Kismet 2019-08-R1 over the repository (`deb https://www.kismetwireless.net/repos/apt/release/buster buster main`).

Btw: In your docs ([https://www.kismetwireless.net/docs/readme/packages/](https://www.kismetwireless.net/docs/readme/packages/)) you write `deb https://www.kismetwireless.net/repos/apt/release/buster stretch main`. That doesn't work for me. I think it's must be `buster`.

Now I get this error:
```
Uncaught exception "deadlock: shared mutex lock not available within 15 (write held)"
Stack trace (most recent call last) in thread 727:
#1    Object "/usr/lib/arm-linux-gnueabihf/libstdc++.so.6.0.25, at 0x7690e5af, in
#0   Object "/usr/bin/kismet, at 0x3b5f37, in TerminationHandler()
Uncaught exception "std:exception"
Uncaught exception "deadlock: shared mutex lock not available within 15 (write held)"
Aborted
FATAL - Capture source did not get PING from Kismet for over 15 seconds; shutting down
```
The error doesn't appears, when I'm at home without walking arround. When I'm outside and walking/driving around I get this error. But not at all the time: Sometimes 3 times during 2 minutes or one time during 15 minutes. Is it possible, that the error appears only if Kismet handle to much devices or packets?
I don't know enough about driver and so on, so I don't understand this error code.
If you need more information, ask!

kismetwireless, 2019-08-08:
If you could use the debug instructions from
https://www.kismetwireless.net/docs/readme/debugging/ I can try to take a
closer look, unfortunately that paste doesn't reveal anything useful.

A rpi3 isn't the optimum hardware; you can definitely over-saturate it; it
can also be dependent on the type of wifi cards you're trying to use.

JK2210, 2019-08-09:
[crash2.log](https://github.com/kismetwireless/kismet/files/3487291/crash2.log)
The file you want from gdb.
I hope I did everything right.
Kismet crashes when Bluetooth and Wi-Fi was started. But not at the beginning.

kismetwireless, 2019-08-09:
I'd also suggest trying the latest nightly, updated this morning. If you've
been following Twitter, it looks like there's a thread contention which is
exacerbated by the insane wifi environment at defcon (but it will show up
elsewhere too, if you're not there).

The latest code this morning reduces packer processing to a single thread
to combat that and seems to be doing it pretty effectively. It's entirely
possible this will alleviate your issue too.

JK2210, 2019-08-11:
I tried the latest nightly today, get the same error. I'm not sure, but it is possible, that this could depends on debian releases? Under Stretch with a kismet version from july I got no errors. Also a problem: Kismet doesn't get gps data over gpsd. but cgps does. This wasn't with the build befor.

JK2210, 2019-08-14:
My SD card reached only 14.9 MB/s. I hope this is enough. I will try your new commit tomorrow! Then this doesn't work I will try an external drive.
I hope this solve my problem.

Thank you for your greate work!

ASoTNetworks, 2019-09-28:
I am having the same issue too on version 2019-09-R1. 

I tried using a USB3.0 flash drive plugged into the Raspberry Pi 3 B a 2.0 port and it will crash. Write speed onto the drive is about 13MB/s.

kismetwireless, 2019-09-28:
You're in an environment with too much data for that platform to keep up,
then; (or your USB drives are also very slow).

You'll need better equipment - in general the pi4 is a much more capable
system if you need something of the same form factor.

If you don't care about capturing data, you could disable packet logging in
the config, otherwise you'll need to get faster equipment for your
environment.

closed 2019-09-30 by JK2210