kismetwireless/kismet #170: Missing kismet-drone when install from repo on kali pi

Nosphorus, 2019-08-09:
Running kali on pi3b, and it seems kismet-drone is missing. I'm trying to set up the Rpi as a kismet drone and eventually tie them all in to a kismet server. When installing from source, the pi3 is incapable of completing the make too.

Is there another way to install the kismet-drone features?

kismetwireless, 2019-08-09:
Remote cap is integrated into the standard capture tools and documented at
https://kismetwireless.net/docs/readme/datasources_remote_capture/

Nosphorus, 2019-08-09:
So has kismet-drone now become obsolete as part of the standard configurations? I'm sure the kismet server had it installed, I'll have to check on Monday.

kismetwireless, 2019-08-09:
Remote cap is present. Kismet drone as a specific program is long long
dead. If you're installing something that installs something called drone,
it's from 2016 or older.

Nosphorus, 2019-08-09:
Ah, weirdly I had found recent-ish documentation that described its use and functionality. Thank you, I'll prep everything new for Monday :)

closed 2019-08-09 by Nosphorus