kismetwireless/kismet #171: Frequency correction in SDR capture?

CharlesCage, 2019-08-13:
Apologies in advance if this can be found anywhere else, but is there currently any way to apply a frequency (or PPM) corrections to Kismet's SDR-based datasources? Having some issues with rtl433 and rtladsb, but discovered that they go away if I select a specific RTL-SDR that I know requires little correction. Read through docs but can't find anything.

kismetwireless, 2019-08-14:
Yep, but the docs hadn't made it from the rtl433 code into the actual
documentation; i've fixed that;

https://www.kismetwireless.net/docs/readme/datasources_sdr_rtl433/

Also it looks like the rtl433 kismet code just didn't respect that option
at all, anyhow; that's also been fixed as of this morning, so it will show
up in tomorrows nightly builds or it's available in git now.

You should be able to pass ppm_error=xxx as part of the source now;

source=rtl433-0:ppm_error=10

and it will get passed to rtl_433 as the '-p' parameter.

Let me know if that's not working after you update; i'm remote right now
and can't easily test.

CharlesCage, 2019-08-15:
Thanks for the quick response and help.

Updated nightly this AM:
`kismet/buster,now 2019+git2019-08-15re799ab12-1 armhf [installed]`

In my /etc/kismet/kismet.conf:
`source=rtladsb-0
source=rtl433-1:ppm_error=25
source=rtl433-2:ppm_error=17
`
From htop after starting kismet:
`rtl_433 -F json -d 2 -f 433.920Mhz
rtl_433 -F json -d 1 -f 433.920Mhz`
(yeah I know I need to set freq for second instance)

Looks like it's still not passing the param.

CharlesCage, 2019-10-07:
Just a reminder that this still exists.