kismetwireless/kismet #172: kismetdb_dump_devices Floating point exception

aabriani, 2019-08-14:
As of **Kismet 2019-08-R1**, trying to use `kismetdb_dump_devices` tool to convert kismetdb to JSON format results in *"Floating point exception"* error.

Here's an example:
```bash
pi@raspberrypi:/ $ sudo kismetdb_dump_devices --in Kismet-20190814-17-42-56-1.kismet  --out - --verbose
* Preparing input database 'Kismet-20190814-17-42-56-1.kismet'...
* Found KismetDB version 6 17 devices
[
Floating point exception
```

aabriani, 2019-08-14:
The same error trying to export it to a JSON file too

kismetwireless, 2019-08-15:
Unfortunately I think I'll need either a copy of that log, or at least a
gdb backtrace of the code, to debug further.

To run it in gdb,

"gdb kismet_dump_devices"
"run <whatever arguments you normally provide."
after it crashes,
"bt"

BTW there's no reason you should need to run the log converter under gdb,
either.

dragorn, 2019-08-15:
I made some guesses w/out the debug info; if it still happens I'll need the
debug logs or the capture log though.

aabriani, 2019-08-15:
Unfortunately, I deleted the log mentioned above yesterday.
But today, strangely, repeating the same operation on new logs didn't result in any errors, and the tool now works fine... Can't explain why.

In case l encounter that error again, I will open a new issue.

Thanks for now, though.

closed 2019-08-15 by aabriani