kismetwireless/kismet #174: New dependency requirements

trevorbryant, 2019-08-21:
Not an unresolvable issue, but more for user resolution knowledge if occured when using custom environments and installing from source.

OS is Ubuntu 19.04.

### Issue

`apt dist-upgrade` is unable to resolve the unmet dependencies from the latest release.

```
The following packages have unmet dependencies:                                                                                                                                                                                               
kismet : Depends: python3-kismetcapturertl433 but it is not installed                                                                                                                                                                        
             Depends: python3-kismetcapturertladsb but it is not installed                                                                                                                                                                       
             Depends: python3-kismetcapturertlamr but it is not installed                                                                                                                                                                        
             Depends: python3-kismetcapturefreaklabszigbee but it is not installed                                                                                                                                                               
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```
`apt --fix-broken install` does not resolve the dependencies and results below.
```
(Reading database ... 329525 files and directories currently installed.)                                                                                                                                                                      
Preparing to unpack .../python3-kismetcapturertl433_9999+git2019-08-21r1b59ef00-1_all.deb ...                                                                                                                                                 
Unpacking python3-kismetcapturertl433 (9999+git2019-08-21r1b59ef00-1) ...                                                                                                                                                                     
dpkg: error processing archive /var/cache/apt/archives/python3-kismetcapturertl433_9999+git2019-08-21r1b59ef00-1_all.deb (--unpack):                                                                                                          
 trying to overwrite '/usr/lib/python3/dist-packages/KismetCaptureRtl433/kismet_cap_sdr_rtl433_mqtt.py', which is also in package python-kismetcapturertl433 9999+git2019-08-20r27f7d4a7-1                                                    
Preparing to unpack .../python3-kismetcapturertladsb_9999+git2019-08-21r1b59ef00-1_all.deb ...                                                                                                                                                
Unpacking python3-kismetcapturertladsb (9999+git2019-08-21r1b59ef00-1) ...                                                                                                                                                                    
dpkg: error processing archive /var/cache/apt/archives/python3-kismetcapturertladsb_9999+git2019-08-21r1b59ef00-1_all.deb (--unpack):                                                                                                         
 trying to overwrite '/usr/lib/python3/dist-packages/KismetCaptureRtladsb-2018.7.1-py3.7.egg-info/SOURCES.txt', which is also in package python-kismetcapturertladsb 9999+git2019-08-20r27f7d4a7-1                                            
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)                                                                                                                                                                          
Preparing to unpack .../python3-kismetcapturertlamr_9999+git2019-08-21r1b59ef00-1_all.deb ...                                                                                                                                                 
Unpacking python3-kismetcapturertlamr (9999+git2019-08-21r1b59ef00-1) ...                                                                                                                                                                     
dpkg: error processing archive /var/cache/apt/archives/python3-kismetcapturertlamr_9999+git2019-08-21r1b59ef00-1_all.deb (--unpack):                                                                                                          
 trying to overwrite '/usr/lib/python3/dist-packages/KismetCaptureRtlamr/__pycache__/__init__.cpython-37.pyc', which is also in package python-kismetcapturertlamr 9999+git2019-08-20r27f7d4a7-1                                              
Preparing to unpack .../python3-kismetcapturefreaklabszigbee_9999+git2019-08-21r1b59ef00-1_all.deb ...                                                                                                                                        
Unpacking python3-kismetcapturefreaklabszigbee (9999+git2019-08-21r1b59ef00-1) ...                                                                                                                                                            
dpkg: error processing archive /var/cache/apt/archives/python3-kismetcapturefreaklabszigbee_9999+git2019-08-21r1b59ef00-1_all.deb (--unpack):                                                                                                 
 trying to overwrite '/usr/lib/python3/dist-packages/KismetCaptureFreaklabsZigbee-2018.7.0-py3.7.egg-info/SOURCES.txt', which is also in package python-kismetcapturefreaklabszigbee 9999+git2019-08-20r27f7d4a7-1                            
No apport report written because MaxReports is reached
Errors were encountered while processing:                                                                                                                                       
 /var/cache/apt/archives/python3-kismetcapturertl433_9999+git2019-08-21r1b59ef00-1_all.deb                                                                                                                                                    
 /var/cache/apt/archives/python3-kismetcapturertladsb_9999+git2019-08-21r1b59ef00-1_all.deb                                                                                                                                                   
 /var/cache/apt/archives/python3-kismetcapturertlamr_9999+git2019-08-21r1b59ef00-1_all.deb                                                                                                                                                    
 /var/cache/apt/archives/python3-kismetcapturefreaklabszigbee_9999+git2019-08-21r1b59ef00-1_all.deb                                                                                                                                           
E: Sub-process /usr/bin/dpkg returned an error code (1)                               
```

### Resolution

```
dpkg -i --force-overwrite /var/cache/apt/archives/python3-kismetcapturertl433_9999+git2019-08-21r1b59ef00-1_all.deb
dpkg -i --force-overwrite /usr/lib/python3/dist-packages/KismetCaptureRtladsb-2018.7.1-py3.7.egg-info/SOURCES.txt
dpkg -i --force-overwrite /var/cache/apt/archives/python3-kismetcapturertladsb_9999+git2019-08-21r1b59ef00-1_all.deb
dpkg -i --force-overwrite /var/cache/apt/archives/python3-kismetcapturertlamr_9999+git2019-08-21r1b59ef00-1_all.deb
dpkg -i --force-overwrite /var/cache/apt/archives/python3-kismetcapturefreaklabszigbee_9999+git2019-08-21r1b59ef00-1_all.deb
```

### Real Resolution
install_pentoo

kismetwireless, 2019-08-21:
Already posted on twitter this morning about it & am rebuilding nightlies
w/ a 'replaces' flag on the python3 packages that ought to resolve that;
but it takes a few hours for the nightlies to build for every platform.

trevorbryant, 2019-08-21:
Aha. Missed that update. Apologies for any dupes issues.

kismetwireless, 2019-08-21:
No problem, good catch and hopefully the new packages fix it seamlessly; I
had to bump them to match the incoming kali package naming for their
packages they're going to add for the new kismet.

dragorn, 2019-08-21:
The new packages should be up now, the python3-xyz packages all flag
python-xyz as replacements.

trevorbryant, 2019-08-21:
Tested clean install and worked great.

Thanks, team.

closed 2019-08-21 by trevorbryant