kismetwireless/kismet #175: dji flight parser broken, can I override it with plugin ?

matanfih, 2019-08-22:
I want to do some experimentation, but would like to do it in a plugin (and not directly interfering a live and kicking open source project !), First thing I noticed playing with my Dji-Sprak is that current DroneID impl seems off
I can't find any example on how to implement proper plugin that will handle packet parsing 
and current examples seems a bit novice for that matter
 
TL;DR
I want to do some experiments for DroneID issue
on the way learn how to write proper plugin 
need help on the later

kismetwireless, 2019-08-22:
You'd be better off updating the stream parser for the DJI protocol, if it
changed.  PRs welcome, it would be much harder to do it as an external
plugin.

More info:

https://github.com/kismetwireless/kismet/blob/master/dot11_parsers/dot11_ie_221_dji_droneid.h
https://github.com/kismetwireless/kismet/blob/master/dot11_parsers/dot11_ie_221_dji_droneid.cc

https://www.kismetwireless.net/docs/devel/dot11_ie_parsers/

If you can provide pcaps of the latest DJI beacons I'd be happy to look.

Also if you're not already aware, if you're using the RC controller, you'll
only be able to see DJI packets with a droneid component if you have an
atheros ath9k pci card, configured for quarter-channel mode; the RC
controller doesn't use standard wifi to talk to the spark.

MAVProxyUser, 2022-03-24:
I'll try to help get this updated soon. The packet format changed slightly. Stop by Mattermost if you want to discuss or contribute further. https://dji-rev.com/dji-rev/channels/town-square

kismetwireless, 2022-03-24:
Awesome!

MAVProxyUser, 2022-04-05:
Parser should work for v2 droneID now. This inherently breaks v1 support (just a little) by providing a false app location that was never broadcast. Other values seem sane. This is a great start. It may be possible to add some logic checks to ensure v1 & v2 can be detected at a later time. https://github.com/kismetwireless/kismet/commit/8a31283dfafb7b085ecb7766256b43d661a9f7ca