kismetwireless/kismet #177: kismet-to-wiglecsv issues

8-bit-oscillations, 2019-08-30:
The kismet-to-wiglecsv script is not outputting the ssid name am I missing a configuration or something as it blank in every line. I can see the data is in the json field in the devices table in the .kismet file if viewed with a sqlite3 viewer.

kismetwireless, 2019-08-30:
Yep; missed that in the conversion to the new SSID aliased field.  Should
be fixed in git now if you want to give it a test; it'll hit the nightlies
by tomorrow morning as well.

8-bit-oscillations, 2019-08-30:
OK thanks

8-bit-oscillations, 2019-08-30:
Ok it's working but it's not working, now all the output is only ssid's none of the clients or lone wifi devices are showing up.
Is it possible to have the same export but where it's a client or a lone device just have the mac address show in the ssid name so you know it's a client?
Thanks......

kismetwireless, 2019-08-30:
Currently no, that's working as designed.  I'll talk to the wigle guys to
see how they'd like to proceed with getting client data (if they're
interested in it).

8-bit-oscillations, 2019-08-30:
I can see it in the sqlite3 file but scripting is not my thing but where "kismet.device.base.name" is null get "kismet.device.base.commonname" because that will just be filled with the mac address.

The device dump script gives too much data which then needs to be trimmed down, 

Thanks for your time.....

kismetwireless, 2019-08-30:
The device script is designed to dump the entire device record as seen by
kismet, so that is performing as designed.

If you want to trim the record you can process the json and pick what
fields you want.

The dump script will always put the entire record out.