kismetwireless/kismet #179: Frequent Crashes. Where is the runtime log?

th3geek, 2019-09-15:
Hi,

I'm getting fatal exceptions every 5-10 minutes while wardriving with 8 adapters. I've been thinking maybe its a bad adapter, but I'm unable to even really look at the errors that kismet is producing since I can't find a runtime logfile. I've looked over the log documentation and don't see anything about a runtime/error log. Can you point me in the correct location?

Also, due to the crashing I have about 25 .kismet database files and would really like to combine them all into one. Even better would be the ability to automatically use the SAME database everytime kismet is run.

Any help is greatly appreciated,
Thanks!

kismetwireless, 2019-09-15:
Output is on the console of the terminal you've run kismet from. If you're
running it automatically from some other system tool you'll need to see
where it logs; systemd logs to syslog.

The likelihood of that many adapters working is pretty dependent on the
drivers and chipset. 8812au adapters absolutely will not work, the driver
is awful and gets worse when there are many adapters.

There is currently no plan to resume an existing log; the logistics of
doing so are too likely to cause significant other problems.

dragorn, 2019-09-15:
if it crashes consistently, try running it under gdb; info on getting a
debug dump is at:

https://www.kismetwireless.net/docs/readme/debugging/

but yeah - if those are 8812au cards, expect exceedingly poor performance
from them, unfortunately.

th3geek, 2019-09-15:
ah ha. Well, I do have (luckily only) two 8812**BU** cards, but I'm guessing that's the culprit. My other cards are all Ralink 5370 (I hope this is a solid card). Can you recommend the few chipsets that perform well with kismet?  I've looked over the kismet website and it seems the only NICs listed there that don't have problems are ZyDAS and RALink rt2x00 based devices?

I didn't have any problems when I originally tested, it wasn't till I was actually driving that it was crashing every 5-10 minutes so I couldn't easily read the screen and with the way kismet outputs to the terminal means there no way to scroll back and see message you may have missed. I guess I'll just start it outputting everything to a file so I can read it later. I'm currently testing to see if it might crash (so far working fine 30 min. in)

Merging the DB wouldn't really be needed if I could `kismet_log_to_pcap *.kismet` but it doesn't seem to like wildcards. Anyone know of a fancy linux command I could use to process all 30+ db files in one command? Not that typing it out 30 times is a big deal, but maybe I'll try to tinker together a small script to do it.

Any input appreciated.

kismetwireless, 2019-09-15:
for x in *.kismet; do something-with-each-file $x; done

If you're on something like a rpi i'd look at your sd card speeds.  Kismet
will absolutely abort if it's gotten more data than it can write w/in a
reasonable timeout, because nothing else can log while the logs are being
synced.  A crummy SD card will kill you as soon as you start getting more
data.