kismetwireless/kismet #180: Wont Compile on Mac OSX

mwarchut-monsoon, 2019-09-18:
Hello.. I downloaded kistmet today and attempted to compile it.  Unfortunately after several minutes it fails with the following error.

ld: warning: ignoring file libkismetdatasource.a, file was built for archive which is not the architecture being linked (x86_64): libkismetdatasource.a
Undefined symbols for architecture x86_64:
  "_adler32_csum", referenced from:
      _probe_callback in capture_kismetdb.c.o
      _open_callback in capture_kismetdb.c.o
  "_cf_find_flag", referenced from:
      _probe_callback in capture_kismetdb.c.o
      _open_callback in capture_kismetdb.c.o
  "_cf_handler_free", referenced from:
      _main in capture_kismetdb.c.o
  "_cf_handler_init", referenced from:
      _main in capture_kismetdb.c.o
  "_cf_handler_loop", referenced from:
      _main in capture_kismetdb.c.o
  "_cf_handler_parse_opts", referenced from:
      _main in capture_kismetdb.c.o
  "_cf_handler_remote_capture", referenced from:
      _main in capture_kismetdb.c.o
  "_cf_handler_set_capture_cb", referenced from:
      _main in capture_kismetdb.c.o
  "_cf_handler_set_open_cb", referenced from:
      _main in capture_kismetdb.c.o
  "_cf_handler_set_probe_cb", referenced from:
      _main in capture_kismetdb.c.o
  "_cf_handler_set_userdata", referenced from:
      _main in capture_kismetdb.c.o
  "_cf_handler_spindown", referenced from:
      _kismetdb_dispatch_packet_cb in capture_kismetdb.c.o
      _kismetdb_dispatch_data_cb in capture_kismetdb.c.o
  "_cf_handler_wait_ringbuffer", referenced from:
      _kismetdb_dispatch_packet_cb in capture_kismetdb.c.o
      _kismetdb_dispatch_data_cb in capture_kismetdb.c.o
  "_cf_params_interface_new", referenced from:
      _probe_callback in capture_kismetdb.c.o
      _open_callback in capture_kismetdb.c.o
  "_cf_parse_interface", referenced from:
      _probe_callback in capture_kismetdb.c.o
      _open_callback in capture_kismetdb.c.o
  "_cf_print_help", referenced from:
      _main in capture_kismetdb.c.o
  "_cf_send_data", referenced from:
      _kismetdb_dispatch_packet_cb in capture_kismetdb.c.o
  "_cf_send_error", referenced from:
      _kismetdb_dispatch_packet_cb in capture_kismetdb.c.o
      _kismetdb_dispatch_data_cb in capture_kismetdb.c.o
      _capture_thread in capture_kismetdb.c.o
  "_cf_send_json", referenced from:
      _kismetdb_dispatch_data_cb in capture_kismetdb.c.o
  "_cf_send_message", referenced from:
      _open_callback in capture_kismetdb.c.o
      _capture_thread in capture_kismetdb.c.o
  "_kismet_datasource__sub_gps__init", referenced from:
      _kismetdb_dispatch_packet_cb in capture_kismetdb.c.o
      _kismetdb_dispatch_data_cb in capture_kismetdb.c.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [kismet_cap_kismetdb] Error 1

Any suggestions... ?

mwarchut-monsoon, 2019-09-20:
No takers on this?

kismetwireless, 2019-09-20:
Can't replicate any errors with latest kismet release or git head on the
only mac system I have available to me; check that you have all the
dependencies installed and any cflags and ldflags set that you need to
point to wherever you installed the deps.

kismetwireless, 2019-09-30:
Can confirm, compiles fine using the process documented at:  https://www.kismetwireless.net/docs/howto/osx/

closed 2019-09-30 by kismetwireless