kismetwireless/kismet #181: Data source 'rtladsb-0' failed to launch

Darkwinder, 2019-09-20:
Hi there,

it's a bit weird, but I am not able to get kismet with rtladsb running as a service. Okay, I know rtladsb support is stilll under heavy development, but I think my problem is a bit more simple...

When I run kismet from the command line as user pi with 

kismet -c rtladsb-0

it works like a charme and I see the planes. When I run kismet from the command line as user pi with just

kismet

it uses of course kismet.conf, which is:

```
source=wlan0mon
source=rtladsb-0
```

The WiFi-Part works perfect no matter if I run kismet from the command line or as a service. But the RTL-Part states:

```
Unable to find driver for 'rtladsb-0'.
Data source 'rtladsb-0' failed to launch
```

My configuration is a RPI and I use the user "pi" to start kismet from the command line. 
Pi is in the group kismet and I rebooted several times. 
I cloned via git, I use the Version 2019-08-GIT and compiled it as suidinstall.

/etc/systemd/system/kismet.service.d/override.conf

is empty, which is just temporary. I initially wrote a user=kismet and group=kismet as shown in the docs, but then I was not able to start the service at all. At the moment I am testing with root and try to use a user account later when it's al up and running.

/lib/systemd/system/kismet.service

```
[Unit]
Description=Kismet
ConditionPathExists=/usr/local/bin/kismet
After=network.target auditd.service

[Service]
User=root
Group=root
Type=simple
ExecStart=/usr/local/bin/kismet --no-ncurses
KillMode=process
TimeoutSec=0
Restart=always

[Install]
WantedBy=multi-user.target
```

The output of rtl_test is:

```
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
```

The output of rtl_adsb is:

```
Found 1 device(s):
   0:  Realtek, RTL2838UHIDIR, SN: 00000001
 
 Using device 0: Generic RTL2832U OEM
 Detached kernel driver
 Found Rafael Micro R820T tuner
 Tuner gain set to automatic.
 Tuned to 1090000000 Hz.
 Exact sample rate is: 2000000.052982 Hz
 Sampling at 2000000 S/s.
 *a8214c422214a578d8b524bc636c;
 *a9010900ac544218ba51816af130;
 *8d4d00c15c67843b191d4ea2c192;

```

WebUI says:

```
* Datasource rtladsb-0 cancelling source probe due to timeout
* Unable to find driver for 'rtladsb-0'. Make sure that any required plugins are loaded, the interface is available, and any required Kismet helper packages are installed.
* Data source 'rtladsb-0' failed to launch: Unable to find driver for 'rtladsb-0'. Make sure that any required plugins are loaded, the interface is available, and any required Kismet helper packages are installed.
```

As I said it must be a very stupid error - any hint is highly appreciated :-)

Dirk

kismetwireless, 2019-09-20:
is rtl_adsb installed in your path?  Are you trying to run it as the same
user as you're running it as in test?

Darkwinder, 2019-09-23:
rtl_adsb is in my path and I successfully tested id with the user "pi" from the command line.

Darkwinder, 2019-09-23:
I recognized today a crash of the kismet service stating "broken pipe", maybe a hardware error. I restarted the kismet service as I did it several times the last days. But this time ADSB is working on the WebUI without any problems. So the configuration was right. I'll check the hardware , but I can close the issue now as the problem must be a local one. Thank you for your hints :-)

closed 2019-09-23 by Darkwinder

avionica70b, 2020-04-27:
I am facing the same problem but could not find its solution. Please help me address this issue