kismetwireless/kismet #186: /alerts/last-time/[TIMESTAMP.UTIMESTAMP]/alerts.itjson

taospartan, 2019-10-23:
Behaviour is not as expected. 
Still returns in ekjson format rather than itjson.

/alerts/all_alerts.itjson and /alerts/all_alerts.json work as expected.

Using the nightly git buld for Debian / Raspbian Buster.

many thanks

Tao

kismetwireless, 2019-10-23:
This is actually expected, but probably not properly documented, behavior.

That format isn't actually compatible with ekjson OR itjson - when you request a specific time, you get back a dictionary with the current timestamp, and a list of alerts; you can't fetch a dictionary as an ekjson or itjson, only vectors.

I'll update ek and it json to flat-out fail and return an error code when fed an incompatible data type, at the least.

taospartan, 2019-10-23:
Sorry for the confusion, was utilising the kismet_rest package to interrogate the alerts and made an assumption from the for loop in the example given.

Many thanks

Tao

kismetwireless, 2019-10-23:
It's a valid assumption, so for now I need to at least throw an error when it's a wrong assumption to make, and add docs for it.  It's an edge case not many people have hit before, i guess.

PR #255 (2020-05-06): alertracker.cc: do not return server timestamp
Do not return server timestamp when using the timestamp alerts API.
This behavior has been added with commit efb29ff and even if it is well
documented, the other timestamp APIs (e.g. device, message, view ...) do
not return a timestamp moreover this timestamp breaks the generation of
ekjson/itjson as both only works on vectors.

Fix #186
Fix kismetwireless/python-kismet-rest#5
commit d12bd07c (2020-05-06): alertracker.cc: do not return server timestamp

Do not return server timestamp when using the timestamp alerts API.
This behavior has been added with commit
efb29ffc80767c2e80e1be707b2d8b3b6b9fce21 and even if it is well
documented, the other timestamp APIs (e.g. device, message, view ...) do
not return a timestamp moreover this timestamp breaks the generation of
ekjson/itjson as both only works on vectors.

Fix #186
Fix https://github.com/kismetwireless/python-kismet-rest/issues/5
closed 2020-05-06 by kismetwireless