kismetwireless/kismet #189: Kismet Wireless Repo Down?

CharlesCage, 2019-11-15:
Hope it's ok to ask about this here. Noticed that the kismet wireless repo appears to be down.

~~~
Err:4 https://www.kismetwireless.net/repos/apt/git/buster buster Release
404  Not Found [IP: 165.227.176.149 443]
~~~

A browser load of https://www.kismetwireless.net/repos/ returned nothing as well.

Again, apologies if this is the wrong place to report, and thanks so much for all your work on this project. I use it regularly, and GREATLY value it.

kismetwireless, 2019-11-15:
Yep; had a power outage and docker got mad.

New nightlies are building as we speak - I spotted it about 30 minutes
before you did; should be back to normal in about 2 hours.

CharlesCage, 2019-11-15:
You're teh awesom.

Thx.

closed 2019-11-15 by CharlesCage