kismetwireless/kismet #193: Push live data to No Sql database

tonybounty, 2019-11-30:
It's possible to write in live datas/alerts/devices... in txt file or to stdout instead of sqlite file ?
I have to put all results in No SQL DB, so I need to parse data from kismet. Sqllite isn't appropriate to do that.

kismetwireless, 2019-11-30:
The kismetdb log contains full JSON records; you can put them through
whatever you want (nosql, elk, whatever you want to format it as), so you
can either process the existing data out of the kismet log into whatever
you want - using for example the python-kismetdb code or the JSON export
code, or you can talk to the kismet rest ui (docs are all on the website).
Either one gives you full JSON records; either from an existing log, or
realtime.

tonybounty, 2019-12-01:
Do you have more concrete example ? I need do push every new data from kismet to ELK in *realtime*, to update my "live dashboard". However, when I lunch Kismet it give me a SQLite 3 file. It's very complicated to parse SQL on every new data because it's a structured file, and this consume lot of resource to request on every new data.
The more convenient way to parse log is to have a simple log file text (in JSON for eg, line by line), then ELK/Prometheus... or other log aggregator can detect and read each new line written.

Can kismet do that ?

taospartan, 2019-12-01:
Have a look at kismet_rest on github, I think that is what you are looking for

tonybounty, 2019-12-01:
So if I understand, if I want to get all the alerts in realtime and push it in ELK, I need to create a bash script with a while loop that content a   

- curl request to http://log:psswd@127.0.0.1:2501/alerts/last-time/$LAST_TIME/alerts.json
- push the JSON result to ELK (log stash)
- update $LAST_TIME+1
- sleep 1 sec.

Am I good ?

taospartan, 2019-12-01:
I’ve done that with python and the kismet_rest package but bash will do it to

tranzmatt, 2020-02-24:
> I’ve done that with python and the kismet_rest package but bash will do it to
> […](#)
> Sent from my iPhone

Would you mind sharing?  I'd like to see how you did it as I need a MongoDB version.