kismetwireless/kismet #196: Remote capture GPS

ardevd, 2019-12-10:
Is remote capture GPS still on the to-do list? I know you can set a fixed position but if I have a GPS receiver plugged into the sensor it would be useful to use it.

kismetwireless, 2019-12-10:
It's on the list, but will require changes to the capture protocol as well
as writing custom gps interfaces for all the supported gps types; i have no
eta and it's currently not under development.  Generally, remote capture is
orthogonal to non-static GPS, because *all* packets are streamed from the
remote source; you generally can't do it with something that doesn't have a
constant connection.

ardevd, 2019-12-10:
You could totally have mobile remote sources with 4G connectivity for example.

alphafox02, 2019-12-28:
Really could have used it in a specific scenario but it’s good to hear it’s still on the to do list.