kismetwireless/kismet #197: Remote capture of multiple protocols

ardevd, 2019-12-10:
What's the recommended way of collecting multiple protocols (WiFi, Bluetooth, SDR) with remote capture? Do I run multiple instances with specific `--source` arguments? Is non Wifi-protocols even supported with remote capture?

Thanks!

kismetwireless, 2019-12-10:
You run multiple capture sources - kismet_cap_linux_wifi for wifi,
kismet_cap_sdr_rtl433 for rtl433 stuff, and so on.  Remote cap should be
supported for every type, and works cross-platform (you can send packets
from an embedded linux to a kismet on macos, for instance).

ardevd, 2019-12-10:
Is there a way to name them? It seems the default behavior is to name them by interface name. Would be useful to provide an arbitrary name myself.

dragorn, 2019-12-10:
yep.  put a name=... in the source definition.

https://www.kismetwireless.net/docs/readme/datasources_remote_capture/

closed 2019-12-16 by ardevd