kismetwireless/kismet #198: Any way to detect connectivity loss with remote capture source?

ardevd, 2019-12-10:
Unless i'm missing something It doesnt seem to be possible to detect when a remote capture data source looses connection. Is there some sort of periodic ping to ensure the source is still reporting data? Would be useful to have if the functionality does not yet exist.

kismetwireless, 2019-12-10:
It should go into error state when it's not connected; there is a constant
ping between kismet and capture sources, both local and remote.  There will
also be messages when it disconnects and reconnects.

ardevd, 2019-12-16:
I can't seem to replicate this behavior. If I poweroff a remote capture source it remains in the data source list and I can't see any alert or message saying the remote source is down. Is this intended behavior?

PR #247 (2020-04-15): kis_datasource.cc: fix error display
Since commit 7ad14b75b7813640ccdd22860a386a9702707fd7, an error is displayed only the first time, for example when loosing the connection with a remote capture.

This seems a bit strange so remove this limitation.

Moreover, errors are not displayed after the source is closed since commit 3bab497c7a70610f0942c14feb2e5cb6ff0d6af4.

This is an issue if the source is closed and then re-opened so remove also this limitation.

Both changes should fix #198
commit 2d7ff774 (2020-04-15): kis_datasource.cc: fix error display

Since commit 7ad14b75b7813640ccdd22860a386a9702707fd7, an error is
displayed only the first time, for example when loosing the connection
with a remote capture.

This seems a bit strange so remove this limitation.

Moreover, errors are not displayed after the source is closed since
commit 3bab497c7a70610f0942c14feb2e5cb6ff0d6af4.

This is an issue if the source is closed and then re-opened so remove
also this limitation.

Both changes should fix #198
closed 2020-04-15 by kismetwireless