kismetwireless/kismet #211: GLIBC  in Debian / Raspbian Buster release not supported

mcshaman, 2020-01-11:
I just tried installing Kismet for the Raspbian Buster release and received the following errors when I started up Kismet.

```
INFO: Probing interface 'mon1' to find datasource type
/usr/bin/kismet_cap_linux_wifi: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.29' not found (required by /usr/bin/kismet_cap_linux_wifi)
INFO: Probing interface 'hci0' to find datasource type
/usr/bin/kismet_cap_linux_wifi: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.29' not found (required by /usr/bin/kismet_cap_linux_wifi)
```

On further inspection it appeared that the latest version available for Raspian Buster is 2.28.

kismetwireless, 2020-01-11:
The images are built on a buster-lite download from the rpi site; not sure
what version you're running - try an update?

dragorn, 2020-01-11:
I'll grab the latest images again and rebuild the build system and see if
that changes anything.

kismetwireless, 2020-01-11:
For what it's worth, here's the package list that gets installed during the
build system preparation:

  RUN echo 'debconf debconf/frontend select Noninteractive' |
debconf-set-selections
  RUN apt-get update && apt-get install -y \
      sudo \
      build-essential \
      git \
      libmicrohttpd-dev \
      pkg-config \
      zlib1g-dev \
      libnl-3-dev \
      libnl-genl-3-dev \
      libcap-dev \
      libpcap-dev \
      libnm-dev \
      libdw-dev \
      libsqlite3-dev \
      libprotobuf-dev \
      libprotobuf-c-dev \
      protobuf-compiler \
      protobuf-c-compiler \
      libsensors4-dev \
      python3 \
      python3-setuptools \
      python3-protobuf \
      python3-pip \
      python3-simplejson \
      librtlsdr0 \
      libusb-1.0-0-dev \
      libubertooth1 \
      libbtbb1 \
      libubertooth-dev \
      libbtbb-dev \
      ruby-dev \
      rubygems1.8 && \
      gem install --no-ri --no-rdoc fpm

I'm rebuilding the build server images from buster-lite-2019-09-26; i'll
see what that changes.

kismetwireless, 2020-01-11:
Test programs seem fine between the build image and a pi3 running
buster-lite-2019-09; firing off a full nightly build.

dragorn, 2020-01-11:
Just tested the latest build against a rpi3 w/ buster-2019-09 and all seems
happy; syncing the new builds now.

mcshaman, 2020-01-11:
Yeah I initially tried the release version. After that failed I tried the nightly build. That worked fine. Thanks for the quick feedback.

kismetwireless, 2020-01-12:
I rebuilt the release with the same parameters.