kismetwireless/kismet #213: Which script to use?

mtnsec, 2020-01-13:
There seems to be two different sets of scripts that convert the kismet db to kml and/or csv.  Which one am I supposed to use?  Which will have the most accurate/best information?

These seem to have been installed alongside of Kismet (I'm using Kali 2019.4)

kismetdb_to_kml
kismetdb_to_wiglecsv

And there is a separate db tools git repo here:  https://github.com/kismetwireless/python-kismet-db

Thanks!

kismetwireless, 2020-01-13:
All depends what you want out of it.

KML is for google earth.

Wigle is for uploading to wigle.

Neither incorporates more than a fragment of the data in the kismetdb log,
since they don't need it to present what they do.

You can dump the entire json record for every device out of the log, or you
can process it yourself with the python-kismetdb module.

https://www.kismetwireless.net/docs/devel/kismetdb/ shows you what's
available in the log.

mtnsec, 2020-01-13:
Thanks!  We're not interested in uploading to Wigle, however the data in the wigle CSV seems more useful than the kismet_log_to_csv output.  Mainly we want to be able to output a human-readable csv that includes SSID, signal strength, encryption types (WPA/WPA2/WEP/OPN, etc.) for documenting (CYA) wireless pentests.

For KML we need to be able to show the location with the strongest signal, again for the purpose of being able to approximate the physical location of the AP as accurately as possible.

I'll check out the db reference, thanks for the link! :)