kismetwireless/kismet #215: Setting up Email for Alerts

MyProjectExpert, 2020-01-14:
Kismet has feature that provide WIDS (Wireless Intrusion Detection System) with alerts.  The best I can determine is that alerts are stored in log files.   

i have a  Pi Zero honeypot that sends emails when my ports are scanned and get early warning of intrusion

Is there a way to have email sent on alerts?    Does this functionality currently exists?

kismetwireless, 2020-01-14:
This doesn't exist - sending email is probably not something that would be
a core part of kismet.

If you run kismet via systemd, it will log output to syslog.  There's also
an example plugin which takes alerts and sends them directly to syslog.

There's also python-kismetrest which will talk to the server and let you
easily extract info, like alerts, and take action on them; I'd suggest
using one of those to do custom alerting.

MyProjectExpert, 2020-01-15:
thank you for suggestions.

q2dg, 2025-10-23:
In addition to what @kismetwireless has said, I would ask, if possible, to create another type of log called "alerts" where all (and only) the alerts would end up. Thanks a lot