kismetwireless/kismet #220: can't regex on target attribute / wanna get all handshakes at once

jeronomy, 2020-02-16:
Hi, howdy ?

Hi ! I'm trying to obtain (with simpler & shorter query as possible), a list of all "Wi-Fi AP" devices for which Kismet got Handshake/s and  or PMKKID.
The goal is to run one query, before shuting down Kismet (and avoiding to parse big database file), and obtain all usefull info & pcap.
(I already visit the modern Kismet Documentation, as well as the legacy.)

I tried  few options:
1. curl -s -b "KISMET=GemmeCookies"  -d "json=%7B%22fields%22%3A+%5B%22kismet.device.base.type%22%2C+%22kismet.device.base.macaddr%22%2C+%22dot11.device.wpa_present_handshake%22%2C+%22dot11.advertisedssid.ssid%22%2C+%22kismet.device.base.last_time%22%5D%7D" 127.0.0.1:2501/devices/last-time/0/devices.ekjson

2. curl -s -b "KISMET=NiceCookies"   -d "{'regex': 'dot11.device/dot11.device.wpa_present_handshake/dot11.wpa_present_handshake', '[^0]+']}"  127.0.0.1:2501/devices/views/all/devices.ekjson 

	=> But unfortunalty, it seems we cannot grep/ regex on other member then ones starting with "kismet." with internal regex of Kismet queries. Am i doing it wrong ? 

Then i tryed to use JQ, to parse JSON output, but it seems JQ don't like parsing JSON with member containing dot into their names (or that i sux with json parser, idk yet)... 

3. curl -s -b "KISMET=$1"  -d "" 127.0.0.1:2501/devices/views/phydot11_accesspoints/devices.json |python -mjson.tool > Wifi-AP-List.json 
	jq "[.[]|{\"dot11.device\""}] Wifi-AP-List.json
	jq "[.[]|{\"dot11.device.wpa_handshake_list\""}] Wifi-AP-List.json
	jq "[.[1]|{\"dot11.device\"}|{\"dot11.device.wpa_handshake_list\"}"] Wifi-AP-List.json

4. The python kismet_rest seems nice, but the doc seems tiny, and from source code it seems we can only query/regex on member starting by "kismet.device.". Amarite?

I finaly manage to do download all pmkid & handshakes, with one bash function, but this is so slow (and painful to read/edit)...
Could you help me find the proper way /cleaner way to do it ?  :)

Another question concerning the Web interface : what would be the best (easier?) way to add a new sortable attribute into the array, like "filter" on number of actual connected client, or others attribute not shown by default,in the Web UI?  
Maybe Python Kismet_rest as new Plugin, or add JS file into Kismet ? The doc ('Extending the WebUI') doesn't seems verbose enough for noobster like me... 
I guess newly created module goes into :  ./http_data/js ; but idk how to link a module to the web ui, to get it loaded automatically...?

Thanks !

dragorn, 2020-02-16:
You can regex on any field, but it needs to be a field that's
understandable as a regex (in other words it's not based on field name,
like you guessed).

I think where you're having trouble is that field is a number, not a string
- you can regex against strings and mac addresses (which are string-like),
but not against numbers.

You can see the type of a field if you look in the prettyjson or in the
tracked_fields page (http://localhost:2501/system/tracked_fields.html); in
this case:

dot11.device.wpa_present_handshake
uint8_t handshake sequences seen (bitmask)

that's a bitmask of which handshakes have been seen, not a string.

Right now there's no good query that will return just those values; you'd
have to grab all access points (such as from the AP view), and then look to
see which ones have handshakes as a non-zero.

Some day there will be a more complex query language - the basics of it are
in place, but nothing usable.  That would let you offload the query to
kismet itself w/ some sort of query language json parameter, it's just not
done yet.  Unfortunately there's no ETA on that right now.

jeronomy, 2020-02-16:
Thx for the answer, now i know :)
I'll try to find a way to organize all snippet for those queries, maybe a notebook, or maybe a dmenu script, idk yet.

Another question, not related: how could we send data in "real time" to the Kismet dashboard ?
For exemple, importing traffic & devices from a remote instance of tshark ?

Is there an api endpoint to import data, or should I create a new alert, and raise it every time i discover a new device (and threat the remote capture as a "virtual interface")?
Or I could  try to suspend Kismet and inject pcap into the dabase directly ?

closed 2020-02-19 by jeronomy