kismetwireless/kismet #221: Kismet won't start - no errors shown

chikko80, 2020-02-16:
Hello, ive been trying to start kismet for 2 days know. I surfed the whole internet but can't find anything that could help. If i start Kismet, that is my output:

INFO: Including sub-config file: /etc/kismet/kismet_httpd.conf
INFO: Including sub-config file: /etc/kismet/kismet_memory.conf
INFO: Including sub-config file: /etc/kismet/kismet_alerts.conf
INFO: Including sub-config file: /etc/kismet/kismet_80211.conf
INFO: Including sub-config file: /etc/kismet/kismet_storage.conf
INFO: Including sub-config file: /etc/kismet/kismet_logging.conf
INFO: Including sub-config file: /etc/kismet/kismet_filter.conf
INFO: Including sub-config file: /etc/kismet/kismet_uav.conf
INFO: More than one override file included; Kismet will process them in 
      the order they were defined.
INFO: Loading config override file '/etc/kismet/kismet_package.conf'
INFO: Optional sub-config file not present: /etc/kismet/kismet_package.conf
INFO: Loading config override file '/etc/kismet/kismet_site.conf'
INFO: Optional sub-config file not present: /etc/kismet/kismet_site.conf

KISMET - Point your browser to http://localhost:2501 (or the address of this system) for the Kismet UI
INFO: Registered PHY handler 'RTL433' as ID 1
INFO: Registered PHY handler 'Z-Wave' as ID 2
INFO: Registered PHY handler 'Bluetooth' as ID 3
INFO: Registered PHY handler 'UAV' as ID 4
INFO: Registered PHY handler 'NrfMousejack' as ID 5
INFO: Registered PHY handler 'BTLE' as ID 6
INFO: Registered PHY handler 'RTLAMR' as ID 7
INFO: Registered PHY handler 'RTLADSB' as ID 8
INFO: Could not open system plugin directory (/usr/lib/arm-linux-gnueabihf/
      kismet/), skipping: No such file or directory
INFO: Did not find a user plugin directory (/root/.kismet//plugins/), 
      skipping: No such file or directory
INFO: GPS track will be logged to the Kismet logfile
ERROR: GPSTRACKER - Failed to find driver for gps type 'true'
LOCAL: This is the first time Kismet has been run as this user.  You will 
       need to set an administrator password before you can use many 
       features of Kismet.  Visit http://localhost:2501/ to configure the 
       password, or consult the Kismet documentation to set a password 
       manually.
INFO: Enabling channel hopping by default on sources which support channel 
      control.
INFO: Setting default channel hop rate to 5/sec
INFO: Enabling channel list splitting on sources which share the same list 
      of channels
INFO: Enabling channel list shuffling to optimize overlaps
INFO: Sources will be re-opened if they encounter an error
INFO: Saving datasources to the Kismet database log every 30 seconds.
INFO: Launching remote capture server on 127.0.0.1:3501
INFO: No data sources defined; Kismet will not capture anything until a 
      source is added.

*** KISMET IS SHUTTING DOWN ***

*** KISMET HAS ENCOUNTERED A FATAL ERROR AND CANNOT CONTINUE.  ***
Shutting down plugins...
WARNING: Kismet changes the configuration of network devices.
         In most cases you will need to restart networking for
         your interface (varies per distribution/OS, but 
         typically one of:
         sudo service networking restart
         sudo /etc/init.d/networking restart
         or
         nmcli device set [device] managed true

Kismet exiting.

Maybe someone already faced this problem and solved it. 

sudo service networking restart
sudo /etc/init.d/networking restart

working fine but don't solve the problem. /etc/NetworkManager/NetworkManager.conf is also managed true.

I dont know how to solve it. It's really demotivating after 2 days of try and always error.

Thanks in advance

kismetwireless, 2020-02-16:
ERROR: GPSTRACKER - Failed to find driver for gps type 'true'

looks like your gps definitions are wrong; check your gps= config
sections.  They should be of the format:

gps=type:options

such as

gps=gpsd:host=localhost,port=2947

chikko80, 2020-02-16:
thanks a lot, with the help of your tip i finally got it going. I also had a few problems with gpsd which have now been resolved. But: I now start my gpsd with the command:

sudo gpsd -n / dev / ttyACM0 -F /var/run/gpsd.sock

and then everything works. However, if I start the gpsd system service via "service gpsd start" then it does not work. Although I have the following settings in / etc / default / gpsd:

START_DAEMON = "true"
USBAUTO = "true"
DEVICES = "/ dev / ttyACM0"
GPSD_OPTIONS = "- n"
GPSD_SOCKET = "/ var / run / gpsd.sock"

Is it normal to run gpsd not via "service xxx start"?

Korhan039, 2020-07-01:
> thanks a lot, with the help of your tip i finally got it going. I also had a few problems with gpsd which have now been resolved. But: I now start my gpsd with the command:
>
> sudo gpsd -n / dev / [...]

Hello firstly ,how did you find the device is this file? ==> / dev / ttyACM0