kismetwireless/kismet #222: No RTL_433 data in Kismet inside Docker

tranzmatt, 2020-02-20:
I'm trying to run kismet inside a Docker instance and am having an issue with RTL_433.  If I run 

rtl_433 -d 1 -f 315M  -F json -v

inside the container, I pick a nearby Toyota

{"time" : "2020-02-20 20:12:28", "model" : "Toyota", "type" : "TPMS", "id" : "f0764eb2", "status" : 128, "pressure_PSI" : 37.000, "temperature_C" : 30.000, "mic" : "CRC"}

But if I try with kismet, I never get anything.  I've tried with two SDR's to make sure.  Same problem.  Works command line, not in kismet. 

My kismet_site.conf contains:

source=rtl433-116112109315:channel=315MHz,name=TPM315

Am I doing something wrong?

tranzmatt, 2020-02-21:
When I run kismet, I don't see the rtl_433 binary running on a "ps ax".

kismetwireless, 2020-02-21:
if I had to make a guess, either rtl433 isn't installed properly in the
docker image, your PATH is broken such that it can't be found by kismet, or
something similar.

I've never tried to run rtl433 inside a docker container, you've introduced
several layers of weirdness there.

tranzmatt, 2020-02-21:
rtl_433 is installed in /usr/local/bin and runs fine w/o setting anything special.  I just added it to the build-kismet.sh in your packaging/docker/Dockerile and did the standard build/install.  I'm using the --privileged switch and I can see the SDR inside via rtl_test.  

Is there a debug option to show it's being called?  I added the -d, but it doesn't give me any insight to kickstarting rtl_433

dragorn, 2020-02-21:
Are you testing with Kismet git?  If not, make sure to - rtl_433 sometime
recently removed one of the command line options and now immediately fails;
this is an option kismet used.  The latest kismet git will no longer call
that option.

If your rtl433 in the docker is new enough to have this change, you'll be
impacted.

tranzmatt, 2020-02-21:
I'm using the kismet-2019-12-R2 branch, but yeah, trying to use the master kismet branch crashes because of some undefined object.

tranzmatt, 2020-02-21:
> Are you testing with Kismet git? If not, make sure to - rtl_433 sometime recently removed one of the command line options and now immediately fails; this is an option kismet used. The latest kismet [...]

Is there a commit you know this happened at on rtl_433?

tranzmatt, 2020-02-25:
Rolling back rtl_433 kismet uses that includes "-G" works.  Perhaps this should be removed from kismet to stay inline with the most recent version?

closed 2020-02-25 by tranzmatt