kismetwireless/kismet #225: Error kis_database

zoben10, 2020-03-09:
I'm running the new Kali 2020. When I run the command : 
$ `kismet -c wlan0mon`

I get this output:

[image]

I install Kismet by running the following commands:
$ `git clone https://www.kismetwireless.net/git/kismet.git`
$ `sudo apt-get install build-essential git libmicrohttpd-dev zlib1g-dev libnl-3-dev libnl-genl-3-dev libcap-dev libpcap-dev libncurses5-dev libnm-dev libdw-dev libsqlite3-dev`
$ `cd kismet`
$ `./configure`
$ `make`
$ `sudo make suidinstall`
$ `sudo usermod -a -G kismet superman`

zoben10, 2020-03-09:
I also get this output when I run the command:
$ `kismet -c wlan0mon`

[image]

DiscoveringWorld, 2020-04-05:
Did you tried with sudo 
sudo kismet -c wlan0mon

DiscoveringWorld, 2020-04-05:
Did you tried with sudo ?
sudo kismet -c wlan0mon

zoben10, 2020-04-09:
Fix my problem by downloading Kali 2019 image instead of the new version. Thank you.