kismetwireless/kismet #227: DEAUTHFLOOD alerts?

ffontaine, 2020-03-10:
It seems `DEAUTHFLOOD` alerts are never sent by kismet. This is not suprising as the only mention of `DEAUTHFLOOD` is in the `README.OLD` and in `kismet_alerts.conf`, the source code does not contain any `activate_configured_alert` for this alert. The only mention of flooding detection that I could found was from 2002: https://github.com/kismetwireless/kismet/commit/2a132d45ef602d8bde5fbe1d5df8c1c2d673606a. Can you give some status on this feature? Was it removed?

kismetwireless, 2020-03-11:
> detection that I could found was from 2002: 2a132d4

Yep looks like it didn't survive some transition in the code.

Patches welcome, otherwise it'll go on the list but no real ETA for now,
sorry.

ffontaine, 2020-03-11:
Thanks for your quick answer, I made a patch that put back the rudimentary mechanism from 2002 (i.e. send an alert if more than 10 deauths are received in one second). I'll send a PR in the next few days. However, to correctly detect flooding, the user will have to set the `packet_dedup_size` to 0. Indeed, in case of flooding, most of the deauth packets will be set as duplicated an so will be dropped before getting parsed. Does this limitation sounds acceptable for you?

commit 89fa9218 (2020-03-19): Put back DEAUTHFLOOD alert

DEAUTH alert will be sent if more than 10 deauthentications or
disassociations are received in one second as done in the
rudimentary mechanism that was added back in 2002:
https://github.com/kismetwireless/kismet/commit/2a132d45ef602d8bde5fbe1d5df8c1c2d673606a

It should be noted that this detection mechanism works better if the
user sets packet_dedup_size to 0 otherwise most of the flooding packets
will be detected as duplicate and so will be discarded.
I would advise to document this limitation.

Fix #227
PR #232 (2020-03-19): Put back DEAUTHFLOOD alert
DEAUTH alert will be sent if more than 10 deauthentications or
disassociations are received in one second as done in the
rudimentary mechanism that was added back in 2002:
https://github.com/kismetwireless/kismet/commit/2a132d45ef602d8bde5fbe1d5df8c1c2d673606a

It should be noted that this detection mechanism works better if the
user sets packet_dedup_size to 0 otherwise most of the flooding packets
will be detected as duplicate and so will be discarded.
I would advise to document this limitation.

Fix #227
closed 2020-03-19 by kismetwireless