kismetwireless/kismet #233: GPS Location Not Present in Alerts

taospartan, 2020-03-18:
When using devicefound no GPS data is given in the alert json. If it doesn't do this by design could this be a feature request? When deploying large numbers of kismet sensors its useful to know the location.

many thanks

Tao

taospartan, 2020-03-20:
Within alerttracker.h, could you explain what this does:

`virtual ~kis_alert_info() {
        if (gps != NULL)
            delete(gps);
    } `

Many thanks

Tao

kismetwireless, 2020-03-21:
That's a virtual C++ destructor which cleans up nested objects on deletion.

https://www.geeksforgeeks.org/virtual-destructor/

taospartan, 2020-03-21:
Many thanks

ffontaine, 2020-03-23:
From my understanding, the issue is that `gps->info` is never set in alertracker.cc. It seems that GPS has been added a few years ago with 46e19e6c52060c46d5f31718eacb8a2a2f603450. However, this feature did not survive the API changes especially 33afdf3e899d76940644855031e06758712b0f01. 

I think the best way to fix it would be to add a `kis_gps_packinfo *gps` parameter to `raise_alert` which means updating all `raise_alert` calls. For each `raise_alert`, the caller will have to specify the gps position to use (i.e. for devicefound, it could be `device->get_location()`).

As I'm a newbie on kismet, I would like to get more feedback from @kismetwireless on this proposal, perhaps there is a cleaner way to do it.

closed 2020-04-03 by kismetwireless

taospartan, 2020-04-07:
Many thanks for the alteration!

Tao

PR #239 (2020-04-03): alertracker: fix GPS location
Currently, location values returned in alerts are always set to 0
because gps->info is never set in alertracker.cc. This feature has been
added a few years ago with 46e19e6. However, it did not survive the API
changes especially 33afdf3.

Fix this by retrieving kis_gps_packinfo from in_pack.

Fix #233