kismetwireless/kismet #235: Segmentation error with  Kismet 2020-03-R1

DiscoveringWorld, 2020-03-30:
Hello,

I got error when I run Kismet for serverals minutes. Kismet crashed with fatal error. I executed with sudo so it's not privilege I think.

Thanks to inform me.

Kismet version: Kismet 2020-03-R1

Error: FATAL:  Error during write(): Broken pipe

Thread 288 "MHD-connection" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xac35c440 (LWP 11293)]
0x002b8c70 in json_adapter::pack(std::ostream&, std::shared_ptr<tracker_element>, std::shared_ptr<std::map<std::shared_ptr<tracker_element>, std::shared_ptr<tracker_element_summary>, std::less<std::shared_ptr<tracker_element> >, std::allocator<std::pair<std::shared_ptr<tracker_element> const, std::shared_ptr<tracker_element_summary> > > > >, bool, unsigned int, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)>) ()

-----------------
(gdb) info threads
  Id   Target Id         Frame
  1    Thread 0xb6ff5150 (LWP 10985) "kismet" 0xb6858674 in select ()
    at ../sysdeps/unix/syscall-template.S:84
  2    Thread 0xb62c8440 (LWP 10986) "kismet" do_sigwait (set=<optimized out>,
    set@entry=0x4d3794, sig=sig@entry=0xb62c7e2c)
    at ../sysdeps/unix/sysv/linux/sigwait.c:64
  3    Thread 0xb5ac7440 (LWP 10987) "msgbus" 0xb6ba49a4 in __pthread_cond_wait
    (cond=0x4f0e60, mutex=0x4f0e48) at pthread_cond_wait.c:188
  4    Thread 0xb52c6440 (LWP 10988) "eventbus" 0xb6ba49a4 in __pthread_cond_wait (cond=0x4f07d8, mutex=0x4f07c0) at pthread_cond_wait.c:188
  5    Thread 0xb48ff440 (LWP 10989) "packethandler" 0xb6ba81a0 in __lll_timedlock_wait (futex=futex@entry=0xb2f34c74, abstime=abstime@entry=0xb48fe4f0,
    private=-1229291192) at lll_timedlock_wait.c:55
  7    Thread 0xb3895440 (LWP 10991) "MHD-listen" clone ()
    at ../sysdeps/unix/sysv/linux/arm/clone.S:61
  8    Thread 0xb2e7e440 (LWP 10993) "timers" 0xb6ba8d60 in nanosleep ()
    at ../sysdeps/unix/syscall-template.S:84
  15   Thread 0xb1b67440 (LWP 11018) "MHD-connection" 0xb6ba4d3c in __pthread_cond_timedwait (cond=0xb1d315c8, mutex=0xb1d315b0, abstime=0xb1b66bd8)
    at pthread_cond_timedwait.c:200
  53   Thread 0xafb63440 (LWP 11056) "MHD-connection" 0xb6ba7f10 in __lll_lock_wait (futex=0xb1b66ad0, private=private@entry=0) at lowlevellock.c:46
  252  Thread 0xab35a440 (LWP 11257) "MHD-connection" 0xb6b9f464 in pthread_join---Type <return> to continue, or q <return> to quit---
 (threadid=<optimized out>, thread_return=0x0) at pthread_join.c:90
  259  Thread 0xaab59440 (LWP 11264) "MHD-connection" 0xb6b9f464 in pthread_join (threadid=<optimized out>, thread_return=0x0) at pthread_join.c:90
  261  Thread 0xb267d440 (LWP 11266) "MHD-connection" 0xb6ba81a0 in __lll_timedlock_wait (futex=futex@entry=0x4fd780, abstime=abstime@entry=0xb267bea8,
    private=-1229291192) at lll_timedlock_wait.c:55
  263  Thread 0xaf362440 (LWP 11268) "MHD-connection" 0xb6b9f464 in pthread_join (threadid=<optimized out>, thread_return=0x0) at pthread_join.c:90
  265  Thread 0xa6b51440 (LWP 11270) "MHD-connection" 0xb6b9f464 in pthread_join (threadid=<optimized out>, thread_return=0x0) at pthread_join.c:90
  266  Thread 0xa07ff440 (LWP 11271) "MHD-connection" 0xb68ddd80 in __udivmoddi4 () from /lib/arm-linux-gnueabihf/libgcc_s.so.1
  267  Thread 0xb0364440 (LWP 11272) "MHD-connection" 0xb6c9f144 in __dynamic_cast () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
  272  Thread 0xa9356440 (LWP 11277) "MHD-connection" 0xb6b9f464 in pthread_join (threadid=<optimized out>, thread_return=0x0) at pthread_join.c:90
  273  Thread 0xa8b55440 (LWP 11278) "MHD-connection" 0xb686cf88 in __GI___clock_gettime (clock_id=1, tp=0xa8b53dd0) at ../sysdeps/unix/clock_gettime.c:99
  274  Thread 0xa8354440 (LWP 11279) "MHD-connection" 0xb6b9f464 in pthread_join (threadid=<optimized out>, thread_return=0x0) at pthread_join.c:90
  277  Thread 0xa7b53440 (LWP 11282) "MHD-connection" 0x0015a614 in local_locker::local_locker(kis_recursive_timed_mutex*) ()
  278  Thread 0xae360440 (LWP 11283) "MHD-connection" __GI___printf_fp_l (
---Type <return> to continue, or q <return> to quit---
    fp=fp@entry=0xae35ea08, loc=0xb68c8d44 <_nl_C_locobj>,
    info=info@entry=0xae35e5dc, args=args@entry=0xae35e5b8) at printf_fp.c:261
  279  Thread 0xb1366440 (LWP 11284) "MHD-connection" 0xb6b9f464 in pthread_join (threadid=<optimized out>, thread_return=0x0) at pthread_join.c:90
  281  Thread 0xa7352440 (LWP 11286) "MHD-connection" 0xb6b9f464 in pthread_join (threadid=<optimized out>, thread_return=0x0) at pthread_join.c:90
  283  Thread 0xa6350440 (LWP 11288) "MHD-connection" 0xb67fb910 in _int_free (
    av=0xb2f00010, p=0xb1dac140, have_lock=0) at malloc.c:4000
  284  Thread 0xa5b4f440 (LWP 11290) "MHD-connection" 0xb6fb9344 in memcmp ()
   from /usr/lib/arm-linux-gnueabihf/libarmmem.so
  285  Thread 0xa21fc440 (LWP 11289) "MHD-connection" 0xb6ba4d3c in __pthread_cond_timedwait (cond=0x609f50, mutex=0x609f38, abstime=0xa21fbbd8)
    at pthread_cond_timedwait.c:200
  286  Thread 0xa534e440 (LWP 11291) "MHD-connection" 0xb686c720 in __lll_lock_wait_private (futex=futex@entry=0xb2f00010) at ./lowlevellock.c:33
  287  Thread 0xaeb61440 (LWP 11292) "MHD-connection" 0xb6ba4d3c in __pthread_cond_timedwait (cond=0xb494c928, mutex=0xb494c910, abstime=0xaeb60bd8)
    at pthread_cond_timedwait.c:200
* 288  Thread 0xac35c440 (LWP 11293) "MHD-connection" 0x002b8c70 in json_adapter::pack(std::ostream&, std::shared_ptr<tracker_element>, std::shared_ptr<std::map<std::shared_ptr<tracker_element>, std::shared_ptr<tracker_element_summary>, std::less<std::shared_ptr<tracker_element> >, std::allocator<std::pair<std::shared_ptr<tracker_element> const, std::shared_ptr<tracker_element_summary> > > > >, bo---Type <return> to continue, or q <return> to quit---
ol, unsigned int, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)>) ()
  289  Thread 0xadb5f440 (LWP 11294) "MHD-connection" 0xb6ba4d3c in __pthread_cond_timedwait (cond=0x62bd20, mutex=0x62bd08, abstime=0xadb5ebd8)
    at pthread_cond_timedwait.c:200
  290  Thread 0xacb5d440 (LWP 11295) "MHD-connection" 0xb6ba81a0 in __lll_timedlock_wait (futex=futex@entry=0x4fd780, abstime=abstime@entry=0xacb5bea0,
    private=-1229291192) at lll_timedlock_wait.c:55
  291  Thread 0xa9b57440 (LWP 11296) "MHD-connection" clone ()
    at ../sysdeps/unix/sysv/linux/arm/clone.S:61
  293  Thread 0xad35e440 (LWP 11298) "MHD-connection" clone ()
    at ../sysdeps/unix/sysv/linux/arm/clone.S:61
  294  Thread 0xb0b65440 (LWP 11299) "MHD-listen" clone ()
    at ../sysdeps/unix/sysv/linux/arm/clone.S:61
(gdb)

kismetwireless, 2020-03-30:
> std::allocator > (std::__cxx11::basic_string<char, std::char_traits,
> std::allocator > const&)>) ()
> pthread_join---Type to continue, or q to quit---
> (threadid=, thread_return=0x0) at pthread_join.c:90
> pthread_join (threadid=, thread_return=0x0) at pthread_join.c:90
> ---Type to continue, or q to quit---
>    std::shared_ptr<tracker_element_summary> > > > >, bo---Type to continue, or
>    q to quit---
>    std::char_traits, std::allocator > (std::__cxx11::basic_string<char,
>    std::char_traits, std::allocator > const&)>) ()

Please try the latest nightlies; if they still crash, please send the
output from the debug instructions:

https://www.kismetwireless.net/docs/readme/debugging/

DiscoveringWorld, 2020-03-31:
I got same error with the released in the debian stretch version  Kismet 2019-09-GIT.

Here is the same error log. For the source of this release I can find that where ? I mean, if you wnat that I compile the code, install with debug and send ou the debug output ?

Thanks to inform me.

Thread 138 "MHD-connection" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x984f4440 (LWP 2316)]
0x002b4d54 in json_adapter::pack(std::ostream&, std::shared_ptr<tracker_element>, std::shared_ptr<std::map<std::shared_ptr<tracker_element>, std::shared_ptr<tracker_element_summary>, std::less<std::shared_ptr<tracker_element> >, std::allocator<std::pair<std::shared_ptr<tracker_element> const, std::shared_ptr<tracker_element_summary> > > > >, bool, unsigned int, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)>) ()

kismetwireless, 2020-03-31:
I need the debug info, and it would really help if you used the 2020
nightlies.  Much has changed from 2019 so I'm not sure where you're getting
those packages.

The directions for the data I need from gdb are available at
https://www.kismetwireless.net/docs/readme/debugging/ .  Nightly git with
debug symbols are available in the official kismet package repos.

Unfortunately that single line you've pasted doesn't tell me anything
useful.

DiscoveringWorld, 2020-04-01:
It,s seems taht the nightlies have the same problem So I will take the source code form the git and compile it with debug. After I will do tests.

DiscoveringWorld, 2020-04-01:
I run the compilation on a Raspberry Pi Zero. It's seems that the compiltion need a lot of ram. I executed make with -j 4  but it's seem to much for the required ram. How many meg or gig  I should put for the swap memory size to avoid that the compilation failed with memory error. .

Thanks to inform me.

dragorn, 2020-04-01:
Debug packages are available on the kismet site, like I mentioned before.

Also a pi0 is just barely, sort of, kind of, capable of running Kismet.
Most likely you're running out of RAM if you're in any sort of busy
environment.

A pi0 is absolutely not suitable as anything other than:
a) a toy to test out
or
b) a remote capture feeding packets to a more capable server.

Don't expect any sort of usable performance out of one.

DiscoveringWorld, 2020-04-01:
Can you tell me the link for the debug packages ? Id did not find anything.

I understand, but I plan to use raspberry py with a battery attached on that do drive or walk outside to do some tests.

The 2016 version without database work perectly. Vut I will ave to do some code to split and merge the infomrations from the xml files. 

I prefert the sqlite3 approach to stored the data. So if not possible, I will do my own coding with the captured files from the 2016 versions.

DiscoveringWorld, 2020-04-01:
Can I compile the Kismet from a  VM with the same kind of processor and create a package for myslef  to use locally with apt get install ? How we can create a package with the compiled source code from make compilation ? Sorry i'm new to linux world ;-)

kismetwireless, 2020-04-01:
The old kismet is completely different and tracked a fraction of the data.
It also is completely incapable of logging a large number of devices while
capturing:  all capture is lost during the creation of the log files, for
instance.

https://kismetwireless.net/docs/readme/packages/

You can cross compile, yes.  It's not trivial and not something I would
want to try to explain in an unrelated bug like this; I suggest googling
"docker and qemu" as a starting point though.

DiscoveringWorld, 2020-04-01:
ok thanks, i did not read well ;-) i did  not see the firts thime the way to install debug package. I will try that and give feedback after.

DiscoveringWorld, 2020-04-01:
Hello,

I got the error.  The details (the full debug log) will be send to your e-mail for debug.

Thread 288 "MHD-connection" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xac35c440 (LWP 11293)]
0x002b8c70 in json_adapter::pack(std::ostream&, std::shared_ptr<tracker_element>, std::shared_ptr<std::map<std::shared_ptr<tracker_element>, std::shared_ptr<tracker_element_summary>, std::less<std::shared_ptr<tracker_element> >, std::allocator<std::pair<std::shared_ptr<tracker_element> const, std::shared_ptr<tracker_element_summary> > > > >, bool, unsigned int, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)>) ()
No inferior number '288'
Undefined command: "".  Try "help".
Undefined command: "0x002b8c70".  Try "help".
#0  0x002b8c70 in json_adapter::pack(std::ostream&, std::shared_ptr<tracker_element>, std::shared_ptr<std::map<std::shared_ptr<tracker_element>, std::shared_ptr<tracker_element_summary>, std::less<std::shared_ptr<tracker_element> >, std::allocator<std::pair<std::shared_ptr<tracker_element> const, std::shared_ptr<tracker_element_summary> > > > >, bool, unsigned int, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)>) ()
#1  0x00131420 in google::protobuf::io::CodedInputStream::ReadLengthAndPushLimit()@plt ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
  Id   Target Id         Frame 
  1    Thread 0xb6ff5150 (LWP 10985) "kismet

kismetwireless, 2020-04-02:
OK.

Unfortunately I don't see any useful info in there that helps pinpoint
anything, so unless I'm able to reproduce it locally or you're able to get
more information, I don't have any ideas.

My strong suspicion is that you're running out of memory on a rpi0; it's
really not suitable for any sort of significant workload.

DiscoveringWorld, 2020-04-02:
I will check the memory with top at the same time to see if it's become 0

DiscoveringWorld, 2020-04-02:
Here are the spec. Maybe it's some data detected throw an error I don't know yet.

I think also it can be the web site at hosted on the rasp at 2501.

Memory free around 28 megs and CPU was with average of 85% of use

I will have to analyse more on my side with rasp and others computers so see what append with my USB external antenna

You can close the issue as unsolved.

Linux raspberrypi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian

512 meg of RAM

1 cpu

DiscoveringWorld, 2020-04-04:
Follow the source code it looks like around this function in the json_adapter.cc

Maybe i should try to replay the file with a reload after crash

void json_adapter::pack(std::ostream &stream, shared_tracker_element e, 
        std::shared_ptr<tracker_element_serializer::rename_map> name_map,
        bool prettyprint, unsigned int depth,
        std::function<std::string (const std::string&)> name_permuter) {

DiscoveringWorld, 2020-04-04:
I got an other error
I sent you the full trace by email.

in  this=<optimized out>) at /usr/include/c++/6/bits/stl_tree.h:1636
1636	      while (__x != 0)

Thread 382 "MHD-connection" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xaeaff440 (LWP 3276)]
0x0021c484 in std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > > >, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > > > >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > > > > >::_M_erase (__x=0xaeb112f8, 
    this=<optimized out>) at /usr/include/c++/6/bits/stl_tree.h:1636
1636	      while (__x != 0)
#0  0x0021c484 in std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > > >, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > > > >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > > > > >::_M_erase (
    __x=0xaeb112f8, this=<optimized out>)

DiscoveringWorld, 2020-04-10:
Helllo, I found a way to compile the code in rasp docker container. After, I zipped the content of the compilation and unzio that to my rasp. I run the code and no problem of segmentation. 
Maybe the problem is in the packaging process  for the deb files ?

DiscoveringWorld, 2020-04-10:
Helllo, I found a way to compile the code in rasp docker container.
After, I zipped the content of the compilation and unzip that to my rp0.

I run the code and no problem of segmentation.

Maybe the problem is in the packaging process for the deb files ?

closed 2020-12-20 by DiscoveringWorld