kismetwireless/kismet #240: SetTrackerValue()  - Undefined symbols during "make" command

mishachandan, 2020-04-05:
I am following the steps in https://www.kismetwireless.net/docs/howto/osx/. 
I see the following error on executing "make" command:
"Undefined symbols for architecture x86_64:
  "void SetTrackerValue<unsigned long>(std::__1::shared_ptr<tracker_element> const&, unsigned long const&)", referenced from:
      dot11_tracked_ssid_group::set_ssid_hash(unsigned long long const&) in phy_80211_ssidtracker.cc.o
  "unsigned long get_tracker_value<unsigned long>(std::__1::shared_ptr<tracker_element> const&)", referenced from:
      dot11_tracked_ssid_group::get_ssid_hash() const in phy_80211_ssidtracker.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [kismet] Error 1"

Environment : MacOS Catalina 10.15.4 
Note : I had issues running "brew install pkg-tool". I am hoping that  installation of Command line tools have solved the issues to pkg-tool.

kismetwireless, 2020-04-05:
> "void SetTrackerValue(std::__1::shared_ptr<tracker_element> const&,
> "unsigned long get_tracker_value(std::__1::shared_ptr<tracker_element>

try the latest git, ought to be solved.

robrant, 2020-04-05:
I think `pkg-tool` is now `pkg-config` for OSX. This post corrects for the instruction on the howto you referenced: https://medium.com/macoclock/kismet-on-macos-catalina-b55e0aa9892d.

mishachandan, 2020-04-06:
It works with cherrypick (#b0add9ff7eecf4f763649aed7169e636284e927f).
_Thanks for quick response and the solution!_

closed 2020-04-06 by mishachandan