kismetwireless/kismet #241: Ubertooth plugin + Kismet : Error on "sudo make plugins-install"

mishachandan, 2020-04-07:
I am referring to install ubertooth plugin https://github.com/greatscottgadgets/ubertooth/wiki/Third-Party-Software. On line "sudo make plugins-install" it throws following error:

`└──╼ $ sudo make plugins-install
PLUGIN-INSTALL: plugin-alertsyslog/
g++ -std=gnu++17 -bundle -flat_namespace -undefined suppress  -shared -rdynamic alertsyslog.cc.o -o alertsyslog.so -ldl -lz -lpcre -lsqlite3 -lstdc++
clang: error: invalid argument '-bundle' not allowed with '-dynamiclib'
make[1]: *** [alertsyslog.so] Error 1
PLUGIN-INSTALL: plugin-dashboard/
mkdir -p //usr/local/lib/kismet/dashboard
/usr/bin/install -c -o "root" -g "wheel" -m 444 manifest.conf //usr/local/lib/kismet/dashboard/manifest.conf
mkdir -p //usr/local/lib/kismet/dashboard/httpd
cp -r httpd/* //usr/local/lib/kismet/dashboard/httpd
PLUGIN-INSTALL: plugin-demo-externalhttp/
mkdir -p //usr/local/lib/kismet/httpproxytest
/usr/bin/install -c -o "root" -g "wheel" -m 444 manifest.conf //usr/local/lib/kismet/httpproxytest/manifest.conf
/usr/bin/install -c -o "root" -g "wheel" -m 555 kismet_proxytest /usr/local/bin/kismet_proxytest;
PLUGIN-INSTALL: plugin-demo-webonly/
mkdir -p //usr/local/lib/kismet/demoweb
/usr/bin/install -c -o "root" -g "wheel" -m 444 manifest.conf //usr/local/lib/kismet/demoweb/manifest.conf
mkdir -p //usr/local/lib/kismet/demoweb/httpd
cp -r httpd/* //usr/local/lib/kismet/demoweb/httpd
PLUGIN-INSTALL: plugin-ubertooth/
g++ -std=gnu++17  -Wall -g -I. -g -O2 -O3 -pthread -I/usr/local/Cellar/protobuf/3.11.4/include -DKS_STR_ENCODING_NONE -I/Users/mc0516/POC/kismet -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c packetsource_ubertooth.cc -o packetsource_ubertooth.o
packetsource_ubertooth.cc:32:10: fatal error: 'packetsource.h' file not found
#include <packetsource.h>
         ^~~~~~~~~~~~~~~~
1 error generated.
make[1]: *** [packetsource_ubertooth.o] Error 1
make: *** [plugins-install] Error 2`

**Environment :**
Mac OS Catalina 10.15.4
Ubertooth - https://github.com/greatscottgadgets/ubertooth/releases/tag/2018-12-R1
Kismet - Latest pull commit#b0add9ff7eecf4f763649aed7169e636284e927f as of Apr 5, 2020

DiscoveringWorld, 2020-04-07:
Its seems like the file is missing from the source repo.

kismetwireless, 2020-04-07:
This plugin is not for any current version of kismet; it would have to be rewritten to use the new API.

The file is not missing - it no longer exists, and will not be returning.

closed 2020-04-07 by kismetwireless