kismetwireless/kismet #242: Startup Errors

kertis8763, 2020-04-07:
Please help me fix the errors
[image]
[image]

DiscoveringWorld, 2020-04-07:
Did you run Kismet with an interface ? Did you set your external card in monitor mode?
Example:
sudo ifconfig wlan1 down
sudo iwconfig wlan1 mode monitor
sudo kismet -c wlan1

kismetwireless, 2020-04-07:
It looks like you installed the proxy test code and are missing some other libraries it expects.  Uninstall that and you'll stop seeing those alerts, but they're not fatal errors and won't affect running kismet.

Putting a card in manual monitor mode has nothing to do with it, and Kismet does that automatically.

kismetwireless, 2020-04-07:
I pushed the latest kismet-external module to pypi; make sure you have that and the latest kismet git and the example plugin will load fine.

closed 2020-04-07 by kismetwireless