kismetwireless/kismet #243: Question about coding for Kistmetwireless

DiscoveringWorld, 2020-04-07:
Hello,

I would like to know, if in the future I would like to do some code on Kismet, how it's work ? I just have to add some feature and do a pull request ?

Do you have any roadmap ? How decide wich feature will be release when ?

Do you work alone on personal time on that or there is a team on that?

Thanks to inform me.

kismetwireless, 2020-04-07:
You emailed me twice AND made an issue within 15 minutes?  Please.  Calm down, and this is NOT an issue to be raised here.

Yes, send a pull request.

Releases are when I feel there are enough features, or important bugs are fixed, or when I coordinate with distribution managers to make sure a release is cut before a major distribution release.

There is no public road map.

closed 2020-04-07 by kismetwireless

DiscoveringWorld, 2020-04-07:
Sorry, i did not know if you will reply. I was trying to fix my message typo but I sent tow messages.