kismetwireless/kismet #250: kismet-2020-04-R2 crashing

tranzmatt, 2020-04-27:
I was running the kismet-2020-04-R2 release to test whether the native rtl433 included TPM capturing (the docs don't say, only about things like remote weather monitors) and it crashed with the following error.  I wasn't using the web ui or anything, just letting it run.  This is on an aarch64 Jetson TX2 device and the previous release I used, kismet-2020-03, would run for days w/o issue.

```
 Stack trace (most recent call last):
 #21   Object "/usr/local/bin/kismet, at 0x55598f860b, in $x
 #20   Object "/lib/aarch64-linux-gnu/libc-2.27.so, at 0x7fb1f086df, in __libc_start_main
 #19   Source "/src/kismet.git/kismet_server.cc", line 970, in main
 #18   Source "/src/kismet.git/pollabletracker.cc", line 126, in select_loop
 #17   Source "/src/kismet.git/pollabletracker.cc", line 165, in process_pollable_select
 #16   Source "/src/kismet.git/pipeclient.cc", line 157, in pollable_poll
 #15   Source "/src/kismet.git/buffer_handler.cc", line 336, in commit_read_buffer_data
 #14   Source "/src/kismet.git/kis_external.cc", line 206, in buffer_available
 #13   Source "/src/kismet.git/datasource_linux_bluetooth.cc", line 39, in dispatch_rx_packet
 #12   Source "/src/kismet.git/kis_datasource.cc", line 672, in __builtin___snprintf_chk
 #11 | Source "/src/kismet.git/kis_datasource.cc", line 1034, in operator()
       Source "/usr/include/c++/7/bits/std_function.h", line 706, in handle_packet_interfaces_report
         703:     {
         704:       if (_M_empty())
         705:    __throw_bad_function_call();
       > 706:       return _M_invoker(_M_functor, std::forward<_ArgTypes>(__args)...);
         707:     }
         708:
         709: #if __cpp_rtti
 #10 | Source "/usr/include/c++/7/bits/std_function.h", line 316, in operator()
     |   314:       _M_invoke(const _Any_data& __functor, _ArgTypes&&... __args)
     |   315:       {
     | > 316:    (*_Base::_M_get_pointer(__functor))(
     |   317:        std::forward<_ArgTypes>(__args)...);
     |   318:       }
       Source "/src/kismet.git/datasourcetracker.cc", line 285, in _M_invoke
 #9    Source "/src/kismet.git/datasourcetracker.cc", line 253, in complete_list
 #8  | Source "/src/kismet.git/datasourcetracker.cc", line 231, in operator()
       Source "/usr/include/c++/7/bits/std_function.h", line 706, in cancel
         703:     {
         704:       if (_M_empty())
         705:    __throw_bad_function_call();
       > 706:       return _M_invoker(_M_functor, std::forward<_ArgTypes>(__args)...);
         707:     }
         708:
         709: #if __cpp_rtti
 #7    Source "/usr/include/c++/7/bits/std_function.h", line 316, in _M_invoke
         313:       static void
         314:       _M_invoke(const _Any_data& __functor, _ArgTypes&&... __args)
         315:       {
       > 316:    (*_Base::_M_get_pointer(__functor))(
         317:        std::forward<_ArgTypes>(__args)...);
         318:       }
         319:     };
 #6    Source "/src/kismet.git/datasourcetracker.cc", line 1012, in operator()
 #5    Source "/src/kismet.git/kis_mutex.h", line 583, in lock
 #4    Source "/src/kismet.git/kis_mutex.h", line 169, in try_lock_for
 #3    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.25, at 0x7fb24bdf67, in __cxa_throw
 #2    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.25, at 0x7fb24bdc9f, in std::terminate()
 #1    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.25, at 0x7fb24bdc53, in
 #0    Source "/src/kismet.git/kismet_server.cc", line 366, in TerminationHandler

```

kismetwireless, 2020-04-27:
If you can run it in debug mode via the directions at
https://www.kismetwireless.net/docs/readme/debugging/ i'll take a look

-m

tranzmatt, 2020-04-27:
```
[Thread 0x7f2affd1c0 (LWP 13708) exited]
[Thread 0x7f2affd1c0 (LWP 13709) exited]
[Thread 0x7fb51431c0 (LWP 13547) exited]
[Thread 0x7fa7fff1c0 (LWP 12132) exited]
terminate called after throwing an instance of 'std::runtime_error'
  what():  deadlock: shared mutex lock not available within 60 (claiming write)

Thread 1 "kismet" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) trace
Tracepoint 1 at 0x7fb77584d8: file ../sysdeps/unix/sysv/linux/raise.c, line 51.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x0000007fb77598b4 in __GI_abort () at abort.c:79
#2  0x0000007fb7cfc10c in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/aarch64-linux-gnu/libstdc++.so.6
#3  0x0000007fb7cf9c54 in ?? () from /usr/lib/aarch64-linux-gnu/libstdc++.so.6
#4  0x0000007fb7cf9ca0 in std::terminate() () from /usr/lib/aarch64-linux-gnu/libstdc++.so.6
#5  0x0000007fb7cf9f68 in __cxa_throw () from /usr/lib/aarch64-linux-gnu/libstdc++.so.6
#6  0x000000555571db24 in kis_recursive_timed_mutex::try_lock_for (this=0x7f78021240, d=...) at kis_mutex.h:169
#7  0x0000005555727f58 in local_locker::local_locker (this=0x7fffffe770, in=...) at kis_mutex.h:429
#8  0x000000555573998c in pipe_client::pollable_poll (this=0x7f78003350, in_rset=..., in_wset=...) at pipeclient.cc:120
#9  0x000000555571f1d0 in pollable_tracker::process_pollable_select (this=this@entry=0x5555b7d680, rset=..., wset=...) at pollabletracker.cc:165
#10 0x000000555571fbfc in pollable_tracker::select_loop (this=0x5555b7d680, spindown_mode=false) at pollabletracker.cc:126
#11 0x00000055556f3fa4 in main (argc=<optimized out>, argv=<optimized out>, envp=<optimized out>) at kismet_server.cc:970
```

dragorn, 2020-04-28:
can you send me the full output from the debug instructions?  Specifically;

info threads
thread apply all bt full

tranzmatt, 2020-04-28:
Trying.. naturally now that I need to do it again... :p

tranzmatt, 2020-04-29:
Finally got something...

[gdb.txt](https://github.com/kismetwireless/kismet/files/4552839/gdb.txt)

kismetwireless, 2020-05-03:
While it's unlikely, if you could try under the latest git or the
2020-04-R3 release, a weird compile loop caused by GCC is fixed.

I don't really think it's related, but it's worth checking.

tranzmatt, 2020-05-07:
I'm doing all this on a Jetson TX2 running the latest bionic distro.   It's GCC is:

gcc (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Does that make a difference?

kismetwireless, 2020-05-07:
Being on a jetson probably has some impact because it seems to be related
to thread timing, and that's a weirder platform where I don't really know
what it does - i've seen more general weird problems coming out of jetson
than just about anything else.  But no, the problem was specifically how
some templates were written in kismet and how gcc handled them vs clang,
but the 2020-04-03 release should address that.

I'm not sure tht code path going wrong could have caused the thread errors,
however.