kismetwireless/kismet #257: kismetdb_to_kml ( or kismetdb_to_gpx) Segmentation fault (core dumped)

jacksnodgrass, 2020-05-10:
I setup kismet on a raspberry pi. 
Kismet 2020-04-R2

If I run 
kismetdb_to_kml  or kismetdb_to_gpx I get: 
Segmentation fault (core dumped)

I have run the kismet_to_* commands ON the Raspberry Pi and on two different Fedora systems running both running Kismet 2020-04-R2. 

I've tried with a dozen different: 
Kismet-20200509-*.kismet 
files. 

I can get: 
kismetdb_dump_devices -f -i Kismet-20200509-*.kismet -o test.file
to run...

-v does not give any useful info. 

Any ideas?

kismetwireless, 2020-05-10:
Should be fixed in 2020-04-R3 released the other week.

jacksnodgrass, 2020-05-14:
I can confirm that this core dump issue was fixed with: 2020-04-R3

closed 2020-05-14 by jacksnodgrass