kismetwireless/kismet #258: Compiler error latest release on macos

loewal, 2020-05-11:
Hi there,
My output of the compilation proces on the code from https://github.com/kismetwireless/kismet.git:
In file included from log_tools/kismetdb_to_wiglecsv.cc:45:
./sqlite3_cpp11.h:526:23: error: no matching constructor for initialization of 'kissqlite3::query_element'
        vec.push_back(query_element{subclause});
I am running macOs Catalina v10.15.5

Other versions give an error not finding symbols for X64...

loewal, 2020-05-11:
This is what I mean...from kismet-master:
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [kismet] Error 1

loewal, 2020-05-11:
So different output from the compiler...because two very different sourcecodes?

kismetwireless, 2020-05-11:
For now just use R2; R3 is a fix for GCC and clang, apparently on every
platform EXCEPT whatever apple is shipping.  I'll take a look when I can,
but if R2 builds for you, just use that one for now since there is no other
difference.

loewal, 2020-05-11:
So I did, cloned R2. The output?

ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [kismet] Error 1
Can't find a solution for this.
Helas!

loewal, 2020-05-11:
My current version is OK, but one always wants some more, insn't it...?
Version:
Kismet 2019-09-GIT

kismetwireless, 2020-05-27:
Should be fixed as of a week or two ago, make sure to do a make clean, especially if you've upgraded something like xcode.

closed 2020-05-27 by kismetwireless