kismetwireless/kismet #259: Compile time errors with clang

lkdevil, 2020-05-11:
Build was successful with gcc but not with clang.

clang version :
```
clang version 9.0.1
Target: aarch64-unknown-linux-android
```
Errors ( trimmed )
```
In file included from log_tools/kismetdb_to_wiglecsv.cc:45:                                     ./sqlite3_cpp11.h:526:23: error: no matching          constructor for initialization of               'kissqlite3::query_element'                       vec.push_back(query_element{subclause});                      ^            ~~~~~~~~~~~  ./sqlite3_cpp11.h:158:12: note: candidate             constructor                                     (the implicit copy constructor) not             viable: no known conversion from 'const         std::list<query_element>' to 'const             kissqlite3::query_element' for 1st              argument                                      struct query_element {                                 ^                                    ./sqlite3_cpp11.h:158:12: note: candidate             constructor                                     (the implicit move constructor) not             viable: no known conversion from 'const         std::list<query_element>' to                    'kissqlite3::query_element' for 1st             argument                                      struct query_element {
```
Some similar issues ( solved ) for reference :
1. https://github.com/STEllAR-GROUP/hpx/issues/2261
2. https://reviews.llvm.org/D22797
3. https://bugs.chromium.org/p/chromium/issues/detail?id=840251

kismetwireless, 2020-05-11:
>    1. STEllAR-GROUP/hpx#2261

Try the current kismet git; I can't recreate on either clang6 or clang10
(tho I don't have a system with clang9 handy)

lkdevil, 2020-05-11:
It works :)

kismetwireless, 2020-05-11:
Cool.  I'll probably do some more tests and push a 2020-05 sooner rather
than later to get those fixes in, even though there's not much by way of
new features.

lkdevil, 2020-05-11:
Great. Issue should be closed or open until you release 2020-05 ?