kismetwireless/kismet #260: Assistance with monitoring setup

DigiAngel, 2020-05-17:
Wow..it's been a while since I used kismet and boy has it changed!  Looks GREAT!.  My goal here is to kind of use kismet to monitor wifi and bluetooth devices in my area, and syslog new devices (after I get a baseline of current devices).  Here's what I have done so far:

1. in kismet_alerts.conf:
~~~
devicefound=00:00:00:00:00:00/00:00:00:00:00:00
devicefound_timeout=2592000
devicelost_timeout=2592000
~~~

2. in kismet_logging.conf:
~~~
kis_log_device_timeout=0
~~~

3. compiled and installed the syslogalert plugin

So far this has worked ok with logs of devices in syslog looking like:
~~~
May 17 08:36:35 devicefound server-ts=1589726195 bssid=<bleh> source=00:00:00:00:00:00 dest=00:00:00:00:00:00 channel= Monitored device <woot> (<meh>) has been found.
~~~

My question is there a better way to do this?  I took a look at the (much different from the console days) log files and it would seem Like I'd need a cron job to extract new device info...not to mention I'm not sure which file?  Thanks for any guidance here...in this day and age I figure would be physical criminals would most likely have a device on them to track.

DigiAngel, 2020-05-25:
Closing due to apparent lack of interest.

closed 2020-05-25 by DigiAngel

kismetwireless, 2020-05-27:
You'd want to hit the REST api with a cron or otherwise scheduled query (ie a process running and sleeping between queries); you can either hit the alerts endpoint w/ a timestamp and get new alerts since the last time you looked, or hit the devices endpoint and query devices directly, fetch devices by MAC, etc.

The rest andpoints are all documented in the dev docs on the kismet site in the documentation category, and there's a kismet-rest python module in pip3 if python is your jam; if not, any language which supports http/https and json would be trivial to talk to the server with, including 'curl' and grep if you were really dedicated.