kismetwireless/kismet #263: No GPS connection in UI, debug says its connected?

markmn123, 2020-05-23:
raspberry pi 4 with USB BU353s4 on raspbian buster light

I can pull info using gpsmon and gpsd without issue

kismet reports GPS connection lost despite it not throwing an error on startup (sometimes it will like this ERROR: GPS device 'serial' encountered a serial error: Serial client 
       closing /dev/ttyUSB0@4800 - connection closed / device removed)

my /etc/default/gpsd
START_DAEMON="true"
USBAUTO="true"
DEVICES="/dev/ttyUSB0"
GPSD_OPTIONS="-b -n"

my /usr/local/etc/kismet.conf (gps portion...wireless works great)
# New GPS configuration
# gps=type:options
#
gps=serial:device=/dev/ttyUSB0
# gps=tcp:host=1.2.3.4,port=4352
# gps=gpsd:host=localhost,port=2947
# gps=virtual:lat=123.45,lon=45.678,alt=1234
# gps=web:name=gpsweb

output of
pi@raspberrypi:~ $ dmesg | grep tty
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M  smsc95xx.macaddr=DC:A6:32:6A:9C:2B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000699] console [tty1] enabled
[    0.397722] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 34, base_baud = 0) is a PL011 rev2
[    4.108988] usb 1-1.4: pl2303 converter now attached to ttyUSB0
[   41.007969] pps pps0: source "/dev/ttyUSB0" added

i have lost alot of sleep over this. any help is greatly appreciated

dragorn, 2020-05-23:
It sounds like you're combining all of those gps statements at once?
That's definitely not what you want to be doing.

If you're using gpsd, use ONLY the gpsd configuration option.  You can not
run GPSD and a serial gps directly controlled by kismet on the same device.

markmn123, 2020-05-23:
Ok Thank you. So i commented out the gps portion of the kismet.conf and alas...
now it says it cant open wlan1 (my usb adapter) even though it was JUST working last night
Does it need to be in monitor mode? i swear it was working without it. either way i tried both in mornitor mode and not in monitor mode and now it wont work
ERROR: Data source 'wlan1' failed to launch, no error provided.
ERROR: Data source 'wlan1 / wlan1' ('wlan1') encountered an error: 
ALERT: SOURCEERROR Source wlan1 has encountered an error. Kismet will 
       attempt to re-open the source in 5 seconds.  (1 failures)

I have it defined in the kismet.conf as well...should i take out wlan1?

markmn123, 2020-05-23:
> Ok Thank you. So i commented out the gps portion of the kismet.conf and alas...
> now it says it cant open wlan1 (my usb adapter) even though it was JUST working last night
> Does it need to be in monitor mode? i swear it was working without it. either way i tried both in mornitor mode and not in monitor mode and now it wont work
> ERROR: Data source 'wlan1' failed to launch, no error provided.
> ERROR: Data source 'wlan1 / wlan1' ('wlan1') encountered an error:
> ALERT: SOURCEERROR Source wlan1 has encountered an error. Kismet will
> attempt to re-open the source in 5 seconds. (1 failures)
> 
> I have it defined in the kismet.conf as well...should i take out wlan1?
> #ran rfkill list and nothing is there
> is there anything that could have disabled it? even though it shows up when i run ifconfig?

markmn123, 2020-05-23:
Ok so even with kismet started and no wlan1 working...i took a look at the web UI and it still says GPS connection lost

markmn123, 2020-05-24:
Closing. Issue resolved and you guys are awesome

closed 2020-05-24 by markmn123