kismetwireless/kismet #264: High stationary device count

markmn123, 2020-05-26:
I hesitated to post but dont know where else to go for help
I got my instance of kismet fully functional a few days ago and have been playing with it ever since.
Its on a Pi4 with awus1900 alfa usb wifi adapter and a bu353-s4 usb gps runing buster light OS

I ran it for about 6-8 hours this weekend on one day and got upwards of 4500+ devices from ONE location. When i hit the ssids tab it was close to 90.
I live in a relatively small neighborhood where i would expect maybe 30-40 ssids (no apartments nearby) and definitely nowhere near 4500 devices. Is this normal?

MyProjectExpert, 2020-05-26:
it sounds a little high to me.  It just depends.  If you have your antenna near front of house where cars pass by, I think you could pick up devices (cell phones) as the pass by.  When I convert and send to WIGLE, most are duplicates and number isn't as high as reported by kismet.  I'm interested to hear what others say..  PS.  connect on Discord.  There is a more activity and conversations about kismet than on GitHub

markmn123, 2020-05-26:
For sure i think i will head to discord

closed 2020-05-26 by markmn123

MyProjectExpert, 2020-05-26:
One other thing to throw at you.  The kismet plugin mobiledashboard (https://github.com/elkentaro/KismetMobileDashboard), has a breakdown the devices as API, Clients, Devices, Bridged, Adhoc, WDS.