kismetwireless/kismet #265: Relative (negative) timestamps don't work with `/devices/views/[VIEWID]`

ntphx, 2020-05-27:
Kismet version 2020-04-R3

When making a request to the device view API using a negative timestamp returns an empty response. 

```
/*
The response to this POST request is empty while it should return all AP devices seen for the last time within the past 10 minutes.
*/

/devices/views/phydot11_accesspoints/last-time/-600/devices.json
```

However, if we make the same request with a positive (epoch) timestamp equal to that of 10 minutes ago the request response returns all devices as expected.

```
// works fine

/devices/views/phydot11_accesspoints/last-time/[EPOCH_10_MINUTES_AGO]/devices.json
```

As a sidenote, request made to `/devices/last-time/[NEGATIVE_TIMESTAMP]/devices.json` work as expected

dragorn, 2020-05-27:
Try again with the git code or the nightly builds; there were a bunch of
fixes around this recently which will go into the next release; I think
this should already be addressed.  Let me know if it isn't!

CharlesCage, 2021-04-03:
Sorry to reactivate, but I'm still running into this problem on 2020-12-R3a. :/

CharlesCage, 2021-04-03:
Also just checked and this is also happening on 9999+git2021-04-03r97687e8e-1 (nightly git)

kismetwireless, 2021-04-04:
Try again with the latest git version.

CharlesCage, 2021-04-05:
@kismetwireless Working with latest git version. :+1:

RFDeck, 2026-11-06:
The URL form from the report works in current code: `GET /devices/views/<view>/last-time/-600/devices.json` turns
the negative value into "now minus 600 seconds" (`devicetracker_view.cc`, `last-time/:timestamp` routes). That
matches the 2021 comment that the git version works.

The POST form of the same endpoint still had the bug. A body like `json={"last_time": -600}` to
`/devices/views/<view>/devices.json` was read into a `uint64_t`, so the negative check never fired and the view
returned every device. #719 fixes that.

If the URL form works for you on 2026-10-R1, this can be closed once #719 is in.