kismetwireless/kismet #266: umlauts for password

c2ho, 2020-05-27:
I was setting up kismet web access for testing reasons using Parrot OS.
Starting it with **_sudo kismet_** resulted in an accessible web session on _**localhost:2501**_ followed by setting up the user account and password.
Messing around with the password and checking the _.conf_ file located at _/root/.kismet/kismet_httpd.conf_ showed me, that you cannot use a password with umlauts (Ö, Ü, Ä) in it. Deleting them from the password inside that .conf file made the login work.

`$kismet --version
 Kismet 2020-04-R2`

Not a serious malfunction but maybe someone wants to use especially these characters for the password. Maybe there's also a reason this isn't working.

kismetwireless, 2020-05-27:
Hmmmm.  I imagine anything that isn't ascii may cause issues - and it
depends how the umlauts are generated (ie if your editor is dropping utf-8
or 16 or whatever...).

I'll see what can be done, but I can definitely see how the parser wouldn't
handle that.

RFDeck, 2026-11-07:
The server side handles this fine: it stores the login as UTF-8 and compares the same bytes, so `curl -u` with a UTF-8
password works on 2026-10-R1. The problem is in the browser. The legacy web UI, which is the default UI in 2026-10-R1,
builds the header with `btoa(user + ":" + pw)`, and that sends ä/ö/ü as Latin-1. The new web UI in master already
encodes the login as UTF-8. #722 does the same for the legacy UI.

closed 2026-11-08 by kismetwireless