kismetwireless/kismet #267: Install via offical website fails

compusolve-rsa, 2020-05-29:
Ubuntu 16
following the instruction here https://www.kismetwireless.net/docs/readme/packages/
results in unmet dependencies

```
The following packages have unmet dependencies:
 kismet : Depends: python-kismetcapturertl433 but it is not installable
          Depends: python-kismetcapturertladsb but it is not installable
          Depends: python-kismetcapturertlamr but it is not installable
          Depends: python-kismetcapturefreaklabszigbee but it is not installable
E: Unable to correct problems, you have held broken packages.
```

dragorn, 2020-05-31:
Should be fixed now, thanks for the heads up

MainakMa, 2024-12-12:
same  issue facing in latest parrot Os same dependencies error any fix or sollution ?

kismetwireless, 2024-12-12:
There is no official Kismet package for Parrot. You can not typically install packages from one distribution on another (in other words, I do not expect any of the existing debs to work on parrot).

I've had reports that the parrot packages (not maintained by me / kismet, and internal to the parrot distro) are incorrect - you'd have to talk to them about that, I have no control over how a distro makes custom packages.

ondrejmo, 2025-04-09:
This issue is still present on Ubuntu 24.04 with the stable version from the official APT repo.

MainakMa, 2025-04-19:
hmm that means parrot definetly flew away in term of packages support

Sechorda, 2025-10-19:
> > This issue is still present on Ubuntu 24.04 with the stable version from the official APT repo.
> 
> Tried to install on Ubuntu 24.04. Same issues.
> 
> Is there a solution please?

Hey peeps yo package is broken

dragorn, 2025-10-19:
> Sechorda left a comment [(kismetwireless/kismet#267)](https://github.com/kismetwireless/kismet/issues/267#issuecomment-3419949236)

Can't replicate with stable or git on 24.04 noble; perhaps you're using the wrong repo.

***@***.***:/# cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=nobleDISTRIB_DESCRIPTION="Ubuntu 24.04 LTS"

***@***.***:/# apt install kismet-capture-linux-bluetooth
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
libcap2 libcap2-bin libnm0 libpam-cap libwebsockets19t64
The following NEW packages will be installed:
kismet-capture-linux-bluetooth libcap2-bin libnm0 libpam-cap libwebsockets19t64
The following packages will be upgraded:
libcap2
1 upgraded, 5 newly installed, 0 to remove and 139 not upgraded.
Need to get 934 kB of archives.
After this operation, 2691 kB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 https://www.kismetwireless.net/repos/apt/release/noble noble/main amd64 kismet-capture-linux-bluetooth amd64 2025-09-R1 [148 kB]
Get:2 http://archive.ubuntu.com/ubuntu noble-updates/main amd64 libcap2 amd64 1:2.66-5ubuntu2.2 [30.0 kB]
Get:3 http://archive.ubuntu.com/ubuntu noble-updates/main amd64 libcap2-bin amd64 1:2.66-5ubuntu2.2 [34.2 kB]
Get:4 http://archive.ubuntu.com/ubuntu noble-updates/main amd64 libpam-cap amd64 1:2.66-5ubuntu2.2 [12.5 kB]
Get:5 http://archive.ubuntu.com/ubuntu noble-updates/main amd64 libnm0 amd64 1.46.0-1ubuntu2.3 [480 kB]
Get:6 http://archive.ubuntu.com/ubuntu noble/universe amd64 libwebsockets19t64 amd64 4.3.3-1.1build3 [229 kB]
Fetched 934 kB in 1s (987 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 44539 files and directories currently installed.)
Preparing to unpack .../libcap2_1%3a2.66-5ubuntu2.2_amd64.deb ...
Unpacking libcap2:amd64 (1:2.66-5ubuntu2.2) over (1:2.66-5ubuntu2) ...
Setting up libcap2:amd64 (1:2.66-5ubuntu2.2) ...Selecting previously unselected package libcap2-bin.

Sechorda, 2025-10-24:
@dragorn any updates on this. can you test the regular kismet package please?

ondrejmo, 2025-10-25:
Hi, I originally (in April) had the same issue on my physical machines with Ubuntu 24.04, but now I was unable to reproduce it in a container.

```bash
podman run -it --rm ubuntu:noble bash
apt update && apt install -y gnupg wget
wget -O - https://www.kismetwireless.net/repos/kismet-release.gpg.key --quiet | gpg --dearmor | tee /usr/share/keyrings/kismet-archive-keyring.gpg >/dev/null
echo 'deb [signed-by=/usr/share/keyrings/kismet-archive-keyring.gpg] https://www.kismetwireless.net/repos/apt/release/noble noble main' | tee /etc/apt/sources.list.d/kismet.list
```

It seems to me that the issue is either triggered by some other package in the system, is intermittent, or has been fixed since (at least on these particular versions).

```
root@3517b055c4fe:/# uname -a
Linux 3517b055c4fe 6.16.12-200.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Oct 12 16:31:16 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
root@3517b055c4fe:/# cat /etc/issue
Ubuntu 24.04.3 LTS \n \l

root@3517b055c4fe:/# dpkg -l | grep kismet
ii  kismet                             2025-09-R1                        amd64        Kismet metapackage
ii  kismet-adsb-icao-data              2025-09-R1                        amd64        Kismet ADSB ICAO data
ii  kismet-capture-antsdr-droneid      2025-09-R1                        amd64        Kismet AntSDR DJI DroneID driver
ii  kismet-capture-bladerf-wiphy       2025-09-R1                        amd64        Kismet BladeRF WiPhy capture helper
ii  kismet-capture-freaklabs-zigbee-v2 2025-09-R1                        amd64        Kismet FreakLabs zigbee hardware capture v2
ii  kismet-capture-hak5-wifi-coconut   2025-09-R1                        amd64        Kismet Hak5 WiFi Coconut capture helper
ii  kismet-capture-linux-bluetooth     2025-09-R1                        amd64        Kismet Linux Bluetooth capture helper
ii  kismet-capture-linux-wifi          2025-09-R1                        amd64        Kismet Linux Wi-Fi capture helper
ii  kismet-capture-nrf-51822           2025-09-R1                        amd64        Kismet NRF51822 BTLE Sniffer capture helper
ii  kismet-capture-nrf-52840           2025-09-R1                        amd64        Kismet NRF52840 BTLE Sniffer capture helper
ii  kismet-capture-nrf-mousejack       2025-09-R1                        amd64        Kismet nRF MouseJack capture helper
ii  kismet-capture-nxp-kw41z           2025-09-R1                        amd64        Kismet NXP KW41Z BTLE and Zigbee Sniffer capture helper
ii  kismet-capture-radiacode-usb       2025-09-R1                        amd64        Kismet Radiacode USB Geiger counter driver
ii  kismet-capture-rtl433-v2           2025-09-R1                        amd64        Kismet rtl_433 wrapper v2
ii  kismet-capture-rtladsb-v2          2025-09-R1                        amd64        Kismet rtlsdr ADSB capture v2
ii  kismet-capture-rz-killerbee        2025-09-R1                        amd64        Kismet Killerbee Sniffer capture helper
ii  kismet-capture-serial-radview      2025-09-R1                        amd64        Kismet Radview geiger counter capture helper
ii  kismet-capture-ti-cc-2531          2025-09-R1                        amd64        Kismet TICC2531 802.15.4 Zigbee Sniffer capture helper
ii  kismet-capture-ti-cc-2540          2025-09-R1                        amd64        Kismet TICC2540 BTLE Sniffer capture helper
ii  kismet-capture-ubertooth-one       2025-09-R1                        amd64        Kismet Ubertooth One BT Sniffer capture helper
ii  kismet-core                        2025-09-R1                        amd64        Kismet core
ii  kismet-logtools                    2025-09-R1                        amd64        Kismet kismetdb log tools (kismetdb)
```