kismetwireless/kismet #268: Previously Selected Columns Not Getting Saved

BeingDatta, 2020-06-03:
All the selected columns are not getting saved have to select them manually everytime which is quite irritating and frustrating.   Plz do something.

Regards.

dragorn, 2020-06-03:
Cannot replicate on any test installs here, with chrome, firefox, or safari.

UI preferences are saved as session data in your browser.  Switching URLs,
browsers, blocking session data, clearing cache, or using temporary/private
browser sessions or a browser which does not store session data/cookies/etc
will prevent them from saving.

MyProjectExpert, 2020-06-04:
Are you clicking on the "Save Changes" on the bottom of screen after making changes?
What version of kismet are you running?  ie kismet -v
Which browser are you using?  Have you tried with other browsers?
Have you tried clearing the cache?
Also try running using chrome in Cognito mode.

BeingDatta, 2020-06-04:
> Are you clicking on the "Save Changes" on the bottom of screen after making changes?
> What version of kismet are you running? ie kismet -v
> Which browser are you using? Have you tried with other browsers?
> Have you tried clearing the cache?
> Also try running using chrome in Cognito mode.

I m using 2020.4 R3, i did save changes at the bottom, i used both firefox and chrome but still not getting saved the chages.

BeingDatta, 2020-06-05:
> Cannot replicate on any test installs here, with chrome, firefox, or safari.
> 
> UI preferences are saved as session data in your browser.  Switching URLs,
> browsers, blocking session data, clearing cache, or using temporary/private
> browser sessions or a browser which does not store session data/cookies/etc
> will prevent them from saving.
> 
> 
> > All the selected columns are not getting saved have to select them
> > manually everytime which is quite irritating and frustrating. Plz do
> > something.
> >
> > Regards.

thanks man you saved my life atlast I found the reason i was using firefox with autoclear data cache history etc on every exit. I guess this is the issue.

Regards.

closed 2020-06-05 by BeingDatta