kismetwireless/kismet #272: UAV monitoring

jhplayer, 2020-06-17:
Hi, I'm trying to scan my UAV/Drone, DJI Mavic but doesn't seem to find it. Kismet is picking up other device data ok.

Is there additional Kismet config I need to do, or is this usually down to the Wifi card?

I'm using a AC1200, 8812AU chipset

Kismet is installed on Ubuntu 18.04 Bionic

Thanks

closed 2020-06-19 by jhplayer

kismetwireless, 2020-06-19:
It all depends exactly what model.

I believe the mavic is RF only - no wifi, I think it's all occusync (or some flavor thereof). You'd need a SDR, and I don't think there are any public implementations of occusync and friends.

Some DJI drones have wifi between the drone and a phone, or between the RC and the phone.  For the ones that have a RC (Spark and Mavic Air, maybe others) the connection between the RC and the drone is 5mhz quarter-channel atheros custom wifi; it's sniffable, but only with an atheros NIC like the 9k.

jhplayer, 2020-06-19:
Thanks for feedback. It's a Mavic Air - will give the 9k a try. Thanks