kismetwireless/kismet #274: Unhandled exception in listinterfaces callback

mezhgano, 2020-06-20:
Hi all

**Kismet 2020-04-R3
5.6.0-kali2-amd64
AWUS036ACH**

Started kismet server, data sources is ok, wi-fi card is running in monitor mode, but 0 packets capture and no data available at all, terminal says:

`  File "/usr/lib/python3/dist-packages/KismetCaptureRtladsb/kismetexternal/__init__.py", line 843, in __handle_kds_listi
nterfaces
    self.listinterfaces(seqno)
  File "/usr/lib/python3/dist-packages/KismetCaptureRtladsb/__init__.py", line 286, in datasource_listinterfaces
    if self.rtlsdr != None:
AttributeError: 'KismetRtladsb' object has no attribute 'rtlsdr'
Unhandled exception in listinterfaces callback
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/KismetCaptureRtlamr/kismetexternal/__init__.py", line 843, in __handle_kds_listin
terfaces
    self.listinterfaces(seqno)
  File "/usr/lib/python3/dist-packages/KismetCaptureRtlamr/__init__.py", line 207, in datasource_listinterfaces
    if self.rtlsdr != None:
AttributeError: 'KismetRtlamr' object has no attribute 'rtlsdr'
Unhandled exception in listinterfaces callback
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/KismetCaptureRtladsb/kismetexternal/__init__.py", line 843, in __handle_kds_listi
nterfaces
    self.listinterfaces(seqno)
  File "/usr/lib/python3/dist-packages/KismetCaptureRtladsb/__init__.py", line 286, in datasource_listinterfaces
    if self.rtlsdr != None:
AttributeError: 'KismetRtladsb' object has no attribute 'rtlsdr'
`
Its some kind of bug or i am doing something wrong?
Sorry, i am completely noob

mezhgano, 2020-06-20:
looks like every time then wi-fi card has capturing the packet, console prompted this error, so its constantly spaming me with this message and i see nothing in GUI

kismetwireless, 2020-06-20:
You can ignore the error, you're missing the rtlsdr libraries & installed
the python rtlsdr components; you can either uninstall those components,
install librtlsdr, or upgrade to a nightly build where I think this was
fixed a while ago but hasn't been integrated into a release yet.

It doesn't impact anything in the UI, it simply prevents discovering any
rtlsdr radios you might have - which wouldn't work anyhow since you don't
have the rtlsdr stuff installed fully.

mezhgano, 2020-06-21:
Ok, thanks for the info 
But, kismet still shows "0" packets captured
Just can't figure out what is going wrong...