kismetwireless/kismet #276: Cannot see ZigBee devices with CC2531

nachorg22, 2020-07-02:
Hi,

I'm using **Kismet 2020-04-R3** version.

I'm trying to monitor ZigBee traffic from a pair of Philips Hue bulbs and its Hub with the TI CC2531 dongle.

The CC2531 device is well detected by Kismet and when I enter the web interface in the "Data Sources" tab I see both the icons and the number of detected packets increase. However, neither through the interface nor by making REST requests with _curl_ I am able to detect any device.

Has anyone else had this problem?
Could it be due to a bug, unfinished functionality, or something I misconfigured?

In contrast, with Wireshark I am able to capture the packets, so I think it is not due to a CC2531 flashing problem.

Thanks in advance.

kismetwireless, 2020-07-02:
Currently there are no decoders for zigbee; packets will be saved in the
pcap and kismetlog files, but there is no phy decoder to present the
interfaces to the UI.

Zigbee presents a significant, special challenge because almost all
networks run in short-address mode, so a 2-byte address with pan 0x00,
making categorizing multiple networks and devices almost impossible.

nachorg22, 2020-07-02:
Okay, thank you very much!

Fantastic job with the whole tool, by the way.

closed 2020-07-02 by nachorg22