kismetwireless/kismet #278: Fields alias not working with Wi-Fi clients API

ffontaine, 2020-07-07:
When passing a two-element array in the fields parameter of the https://www.kismetwireless.net/docs/devel/webui_rest/phy80211/#wi-fi-clients API, kismet does not return correctly the alias as it should be (https://www.kismetwireless.net/docs/devel/webui_rest/commands/#field-specifications).

For example:

```
POST /phy/phy80211/clients-of/4202770D00000000_5031C6DA20800000/clients.itjson HTTP/1.1
json={"fields":[["kismet.device.base.macaddr","macaddr"]]}
```

returns:

`{"kismet.device.base.macaddr": "3C:18:A0:76:3A:7B"}`

instead of: 

`{"macaddr": "3C:18:A0:76:3A:7B"}`

It should be noted that alias is working perfectly with https://www.kismetwireless.net/docs/devel/webui_rest/devices/#devices-by-phy-type. I'm not sufficiently skilled to fix the issue myself as I've not even found where fields are managed in the source code?

kismetwireless, 2020-07-07:
I'll have to dig into it; it's possible that the summarization step isn't
done at all for the itjson process; I honestly forget.

Field aliasing and renaming is done via the summarization calls which
transform the data before the serialization call; in the older code this is
in the various endpoints throughout the codebase, in the newer code it's
automated and pulled into the simplified post/etc endpoints with
callbacks.

It's possible too that it's broken in one of those (newer vs older) methods
and only showing up on one of them; I'll try to take a look.  The original
code used more polymorphism to make a handler, while the new code aims for
greater simplicity by making the entire handler an object w/ std::function
methods so that there are fewer weird dependencies.

-m

ffontaine, 2020-07-07:
Thanks for your feedback, I think the issue is not related to itjson as this works for http://localhost:2501/devices/views/phydot11_accesspoints/devices.itjson. I'll also try to check what could go wrong on my side.

closed 2020-07-11 by kismetwireless

dragorn, 2020-07-11:
After many hours of debugging and chasing code paths, turns out I simply
had forgotten to *pass the calculated rename map to the serializer* at the
final stage.

So there's that.

Fixed in git, will go into the nightlies.  Was actually broken for any
endpoint using the new internal endpoint API which was (supposed to) make
the code simpler to maintain.

ffontaine, 2020-07-11:
Thanks for fixing this issue

commit 82c28cca (2020-07-11): Simplify rename-from-json function, and actually use rename maps

Make the rename-from-json common function a little simpler and identical
to other instances of the same code.

Actually pass the rename map to the serialization system, fixing the
rename bug.

Fixes #278