kismetwireless/kismet #281: Make fails on kali-linux-2020.2b-rpi3-nexmon-64 in kis_net_microhttpd.cc

darkgrue, 2020-07-12:
Trying to compile on a Pi 4 running Kali (Linux kali 4.19.118-Re4son-v8l+ #1 SMP PREEMPT Thu May 7 02:54:03 UTC 2020 aarch64 GNU/Linux). Everything's up to date, and I've followed the Kismet Compiling Quickstart (as well as double-checked the list of packages referenced in https://github.com/kismetwireless/kismet/issues/211#issuecomment-573330638). I have 4GB of RAM on the Pi and have swap enabled as well - shouldn't be running out of memory during the build (which seems to be a problem on the Pi in general).

When I compile, it's dying in kis_net_microhttpd.cc:

```
kali@kali:/usr/local/src/kismet$ sudo make -j$(nproc)
g++ -std=gnu++17 -Wall -g -I. -g -O2 -O3 -pthread -DKS_STR_ENCODING_NONE  -c kis_net_microhttpd.cc -o kis_net_microhttpd.cc.o
g++ -std=gnu++17 -Wall -g -I. -g -O2 -O3 -pthread -DKS_STR_ENCODING_NONE  -c system_monitor.cc -o system_monitor.cc.o
g++ -std=gnu++17 -Wall -g -I. -g -O2 -O3 -pthread -DKS_STR_ENCODING_NONE  -c base64.cc -o base64.cc.o
g++ -std=gnu++17 -Wall -g -I. -g -O2 -O3 -pthread -DKS_STR_ENCODING_NONE  -c kis_httpd_websession.cc -o kis_httpd_websession.cc.o
g++ -std=gnu++17 -Wall -g -I. -g -O2 -O3 -pthread -DKS_STR_ENCODING_NONE  -c kis_httpd_registry.cc -o kis_httpd_registry.cc.o
kis_net_microhttpd.cc: In member function 'int kis_net_httpd::start_httpd()':
kis_net_microhttpd.cc:471:17: error: invalid conversion from 'int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, size_t*, void**)' {aka 'int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, long unsigned int*, void**)'} to 'MHD_AccessHandlerCallback' {aka 'MHD_Result (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, long unsigned int*, void**)'} [-fpermissive]
  471 |                 &http_request_handler, this,
      |                 ^~~~~~~~~~~~~~~~~~~~~
      |                 |
      |                 int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, size_t*, void**) {aka int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, long unsigned int*, void**)}
In file included from kis_net_microhttpd.cc:31:
/usr/include/microhttpd.h:2428:45: note:   initializing argument 5 of 'MHD_Daemon* MHD_start_daemon(unsigned int, uint16_t, MHD_AcceptPolicyCallback, void*, MHD_AccessHandlerCallback, void*, ...)'
 2428 |                   MHD_AccessHandlerCallback dh, void *dh_cls,
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~
kis_net_microhttpd.cc:477:40: error: invalid conversion from 'int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, size_t*, void**)' {aka 'int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, long unsigned int*, void**)'} to 'MHD_AccessHandlerCallback' {aka 'MHD_Result (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, long unsigned int*, void**)'} [-fpermissive]
  477 |                 http_port, NULL, NULL, &http_request_handler, this,
      |                                        ^~~~~~~~~~~~~~~~~~~~~
      |                                        |
      |                                        int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, size_t*, void**) {aka int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, long unsigned int*, void**)}
In file included from kis_net_microhttpd.cc:31:
/usr/include/microhttpd.h:2428:45: note:   initializing argument 5 of 'MHD_Daemon* MHD_start_daemon(unsigned int, uint16_t, MHD_AcceptPolicyCallback, void*, MHD_AccessHandlerCallback, void*, ...)'
 2428 |                   MHD_AccessHandlerCallback dh, void *dh_cls,
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~
kis_net_microhttpd.cc: In static member function 'static int kis_net_httpd::http_request_handler(void*, MHD_Connection*, const char*, const char*, const char*, const char*, size_t*, void**)':
kis_net_microhttpd.cc:873:35: error: invalid conversion from 'int (*)(void*, MHD_ValueKind, const char*, const char*, const char*, const char*, const char*, uint64_t, size_t)' {aka 'int (*)(void*, MHD_ValueKind, const char*, const char*, const char*, const char*, const char*, long unsigned int, long unsigned int)'} to 'MHD_PostDataIterator' {aka 'MHD_Result (*)(void*, MHD_ValueKind, const char*, const char*, const char*, const char*, const char*, long unsigned int, long unsigned int)'} [-fpermissive]
  873 |                         kishttpd->http_post_handler, (void *) concls);
      |                         ~~~~~~~~~~^~~~~~~~~~~~~~~~~
      |                                   |
      |                                   int (*)(void*, MHD_ValueKind, const char*, const char*, const char*, const char*, const char*, uint64_t, size_t) {aka int (*)(void*, MHD_ValueKind, const char*, const char*, const char*, const char*, const char*, long unsigned int, long unsigned int)}
In file included from kis_net_microhttpd.cc:31:
/usr/include/microhttpd.h:3465:49: note:   initializing argument 3 of 'MHD_PostProcessor* MHD_create_post_processor(MHD_Connection*, size_t, MHD_PostDataIterator, void*)'
 3465 |                            MHD_PostDataIterator iter, void *iter_cls);
      |                            ~~~~~~~~~~~~~~~~~~~~~^~~~
kis_net_microhttpd.cc:947:26: error: invalid user-defined conversion from 'kis_net_httpd::http_request_handler(void*, MHD_Connection*, const char*, const char*, const char*, const char*, size_t*, void**)::<lambda(void*, MHD_ValueKind, const char*, const char*)>' to 'MHD_KeyValueIterator' {aka 'MHD_Result (*)(void*, MHD_ValueKind, const char*, const char*)'} [-fpermissive]
  947 |                 }, concls);
      |                          ^
kis_net_microhttpd.cc:938:17: note: candidate is: 'constexpr kis_net_httpd::http_request_handler(void*, MHD_Connection*, const char*, const char*, const char*, const char*, size_t*, void**)::<lambda(void*, MHD_ValueKind, const char*, const char*)>::operator int (*)(void*, MHD_ValueKind, const char*, const char*)() const' <near match>
  938 |                 [](void *cls, enum MHD_ValueKind, const char *key, const char *value) -> int {
      |                 ^
kis_net_microhttpd.cc:938:17: note:   no known conversion from 'int (*)(void*, MHD_ValueKind, const char*, const char*)' to 'MHD_KeyValueIterator' {aka 'MHD_Result (*)(void*, MHD_ValueKind, const char*, const char*)'}
In file included from kis_net_microhttpd.cc:31:
/usr/include/microhttpd.h:2699:49: note:   initializing argument 3 of 'int MHD_get_connection_values(MHD_Connection*, MHD_ValueKind, MHD_KeyValueIterator, void*)'
 2699 |                            MHD_KeyValueIterator iterator,
      |                            ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
make: *** [Makefile:730: kis_net_microhttpd.cc.o] Error 1
make: *** Waiting for unfinished jobs....
```

Not sure how to interpret the error in terms of what I need to do next. Is it related to attempting to build in a 64-bit environment? Suggestions or additional information I can provide?

k3an3, 2020-07-12:
I am also encountering the same issue on the latest Kali release on an amd64 machine. The solution was to download and build libmicrohttpd-0.9.51.

ffontaine, 2020-07-12:
The issue is indeed related to libmicrohttpd. Since version 0.9.71, callbacks must return an enum instead of an int. All packages using libmicrohttpd have to be updated (e.g. https://github.com/systemd/systemd/commit/d17eabb1052e7c8c432331a7a782845e36164f01).

Here is an extract of https://lists.gnu.org/archive/html/libmicrohttpd/2020-06/msg00013.html:
 
> Furthermore, the release introduces an 'enum MHD_Result' instead of
> #defines for MHD_YES/MHD_NO. This is intended to make it easier to check
> for certain API misuse bugs by providing better types (not everything is
> an 'int').  While this does NOT change the binary API, this change
> _will_ cause compiler warnings for all legacy code -- until 'int' is
> replaced with 'enum MHD_Result'.
> 
> If you want your code to build without warnings on both older and newer
> MHD releases, you may want to introduce a MHD_RESULT as done here:
> 
> https://git.gnunet.org/gnunet.git/tree/src/include/gnunet_mhd_compat.h

darkgrue, 2020-07-12:
So short term, I have to revert my version of libmicrohttpd (which I see is 0.9.71-1, so it'd be subject to the new parameter typing, and matches the errors I'm getting). Longer-term, Kismet needs some changes in it's code to account for the changes in libmicrohttpd post-version 0.9.71?

ffontaine, 2020-07-12:
Yes, that's it. I didn't check it but I assume the change in kismet is not so big (as can be seen in systemd commit), you could even try to patch it yourself and send a pull request. It could be faster than reverting libmicrohttpd.

kismetwireless, 2020-07-12:
The changes are going to be very broad, if not deep; the microhttpd team
really screwed anyone using C++ with their library, to the point that if I
had any alternatives I'd be pretty tempted at looking at them, instead.
~300 line diff so far and it's not even close to being compileable,
probably going to be a thousand line change touching a dozen files.

darkgrue, 2020-07-12:
> The changes are going to be very broad, if not deep; the microhttpd team really screwed anyone using C++ with their library, to the point that if I had any alternatives I'd be pretty tempted at [...]

Ooof. That's unfortunate. Well then, I'll work on reverting my version of libmicrohttpd, and wait patiently for Kismet changes (libmicrohttpd or otherwise).

Thank you all for the help, at least this should get me up and running!

commit c4ec91b8 (2020-07-13): Workaround arbitrary API change in libmicrohttpd 0.97.1

libmicrohttpd introduced a c++-breaking change arbitrarily in a point
release, making kismet uncompileable; rewrite every http endpoint
handler to now use an enum on 0.97.1 and an int on previous versions.

Fixes #281
closed 2020-07-13 by kismetwireless