kismetwireless/kismet #283: Kismet 2020-04-R2 kismetdb_to_wiglecsv Segmentation fault

Czechball, 2020-07-18:
Hello,

when I try to convert my Kismet log to Wigle.net csv, I get the following:

```
* Preparing input database 'Kismet-20200718-11-43-40-1.kismet'...
* Found KismetDB version 6
* Found 292 devices, 131288 usable packets, 150531 total packets
* Starting to process file, max device cache 1000
Segmentation fault (core dumped)
```
My Kismet version is 2020-04-R2.

System info:
```
OS: Manjaro 20.0.3 Lysia
Kernel: x86_64 Linux 5.6.16-1-MANJARO
CPU: Intel Core i7-5600U
RAM: 15885MiB
```

Kismet was started manually via the `kismet` command. GPS was configured as static virtual. Kismet was then stopped with Ctrl+C in terminal.

kismetwireless, 2020-07-18:
Should already be fixed in -r3

Czechball, 2020-07-18:
Confirmed. Upgraded to r3 and issue is fixed

closed 2020-07-18 by Czechball