kismetwireless/kismet #288: cross compiling kismet (for arm) from source fails on debian buster

junaidnaseer, 2020-08-08:
So, I downloaded kismet from github and installed the dependencies listed there including `  #zlib1g-dev`!

Unfortunately, no matter what I try (reinstalling that package, e.g.), the autoconf/configure fails with the error:

> checking for unwind.h... yes
> checking execinfo.h usability... yes
> checking execinfo.h presence... yes
> checking for execinfo.h... yes
> checking for stdint.h... (cached) yes
> checking for strerror_r() return... char *
> checking for accept() addrlen type... socklen_t
> checking for pipe2... yes
> checking for special C compiler options needed for large files... no
> checking for _FILE_OFFSET_BITS value needed for large files... 64
> checking for dlopen in -ldl... yes
> checking for deflate in -lz... no
> configure: error: libz is required and could not be found

Strangely, this error does NOT pop up, if I do NOT cross-compile.

It could be related to [this issue](https://github.com/kismetwireless/kismet/issues/22).

kismetwireless, 2020-08-08:
You need to install all the dependent libraries as the arm versions, inside
your cross compiling environment; so you need to get libz for arm inside
whatever your crossbuild environment is (you'll also need a dozen other
libraries built inside your crossbuild env).

Generally you can't just install packages for crossbuild - those are the
packages for your host system, not the target build environment, so you
usually would need to manually fulfill all the deps in your build env.

The kismet packages are built using docker images and emulating arm via
qemu for this reason; it can be quite a mess setting up an exact crossbuild
env.

junaidnaseer, 2020-08-08:
omg, feeling like a noob. Ok, I guess I should delete this issue or something. Actually, I was trying to cross-compile using lxc but that there were some problems with lxc. I try with qemu. Thanks for the response.

closed 2020-08-08 by junaidnaseer

junaidnaseer, 2020-08-30:
I just wanted to give a little follow up feedback. I managed to cross compile kismet for raspberry-pi (arm) using qemu. Thanks for the hint. I should've seen the notice mentioned on the kismet wireless page. All seems to run well for the moment. :+1: