kismetwireless/kismet #291: Broken clients-of API with fields on master

ffontaine, 2020-08-17:
The clients-of API (https://www.kismetwireless.net/docs/devel/webui_rest/phy80211/#wi-fi-clients) is broken when using the following fields:
```
        [['kismet.device.base.key', 'key'],        
        ['kismet.device.base.manuf', 'manufacturer']]
```

The following error is displayed by kismet:

`Aug 17 2020 17:42:54 HTTPD: Uncaught exception 'invalid trackedelement access id -1, cannot use a vector[x] as a map[field, x]' on '/phy/phy80211/clients-of/4202770D00000000_8F0F5532A6DC/clients.itjson'`

And the endpoint returns an empty result.
Reverting cea6e93bbff8e525fb97fd1cca36972469a2b859 fix the issue.

kismetwireless, 2020-08-17:
Try the latest git; I didn't explicitly fix this, but I fixed other things
and can't recreate it locally, i get proper client lists for itjson, json,
prettyjson, etc.

ffontaine, 2020-08-18:
I can reproduce it with latest git.
It should be noted that I'm using python-kismet-rest to send the following command:

`2020-08-18 09:25:14,723 DEBUG kismet_rest.logger interact_yield: POST against http://localhost:2501/phy/phy80211/clients-of/4202770D00000000_5031C6DA2080/clients.itjson with {'json': '{"fields": [["kismet.device.base.macaddr", "macaddr"]]}'}`

I'll try to understand the issue but the "invalid trackedelement access id -1" is suspicious.

dragorn, 2020-08-18:
I'll take a look.  The -1 is normal; that's nothing to be concerned about -
it's a dynamic top-level field which doesn't need a registered ID.

closed 2020-08-18 by kismetwireless

ffontaine, 2020-08-18:
Thanks :+1:

commit 7fae8f14 (2020-08-18): Add dispatcher for generic tracked elements that made it to summary

Add dispatcher for maps and vectors that made it to the final summary
stage w/out an explicit type

Fixes #291