kismetwireless/kismet #298: Safari loses connection.

ehogeweg, 2020-08-27:
Kismet Browser UI keeps losing connection. Apparently because of a permission issue. Kismet engine runs fine

macOS Catalina.
Tested with Safari and Chrome. Normal mode and private browsing mode.
Running kismet with 'sudo kismet -c ticc2540'
Kismet version 2020-00-GIT as of today (8/26)

Tried running Safari as root but that didn't seem to make a difference.

[image]
[image]

kismetwireless, 2020-08-27:
You're probably using the latest libmicrohttpd, which is broken on osx.
Check the compile guide in the readme, you need to manually backlevel the
version brew uses.

As far as I can tell this is a microhttpd error as even their examples
exhibit the same problem.

ehogeweg, 2020-08-27:
Roger. Thanks for the quick response.

closed 2020-08-27 by ehogeweg